micro-manager/micro-manager #1891: AcqEngJ / MultiMDA not working when autofocus is enabled

nicost, 2024-02-14:
From image.sc (https://forum.image.sc/t/help-solicited-to-test-new-micro-manager-features/85667/9): Any update on the multi MDA not working if autofocus is enabled ?

I can confirm I experienced the same inconvenience: the multi MDA doesn’t start (more precisely, individual windows for MDA appear but the microscope stays still and no image or anything is captured) if autofocus is enabled in the multi MDA settings ,irrespectively of it being enabled or not in the indivudal MDAs.

PS: i forgot to mention that we rely on a hardware autofocus device (Nikon PFS)

ilyasdc, 2024-02-15:
According [the comment by Thomas Julou](https://forum.image.sc/t/help-solicited-to-test-new-micro-manager-features/85667/6?u=id-c), it doesn't matter which autofocus is used, even software one are failing.

I wonder if it's not a race condition while trying to acquire the lock on the autofocus "device" ?

nicost, 2024-02-15:
Not sure, there is no problem when using the Demo Configuration (i.e., select the new Acquisition Engine under options, run an MDA with Autofocus checked, and see the MDA succeeding).  Will need to test with actual hardware.

nicost, 2024-02-17:
Just fixed a divide by zero exception that I encountered with the demo config (#1893).  Wondering if that also fixes this issue.  I would really be helped with the relevant logs (you can find these in Corelogs find the file with the correct date).

nicost, 2024-02-17:
This was not supposed to be closed by that PR.

ilyasdc, 2024-02-18:
> Just fixed a divide by zero exception that I encountered with the demo config (#1893). Wondering if that also fixes this issue. I would really be helped with the relevant logs (you can find these in Corelogs find the file with the correct date).

Here are two log files, one with PFS, the other with software (oughtafocus) autofocus:
[CoreLog20240218T161426_pid11152_pfs.txt](https://github.com/micro-manager/micro-manager/files/14323335/CoreLog20240218T161426_pid11152_pfs.txt)
[CoreLog20240218T161646_pid10620_oughtafocus.txt](https://github.com/micro-manager/micro-manager/files/14323336/CoreLog20240218T161646_pid10620_oughtafocus.txt)

It seems that the acquisition is immediately aborted: 

```2024-02-18T16:14:41.273880 tid13216 [dbg,dev:COM3] SetCommand -> CSS?\r
2024-02-18T16:14:41.279077 tid7720 [IFO,App] got event: x 12095.0y  -23662.5	channel: 0	time: 0	position: 0
2024-02-18T16:14:41.280068 tid5452 [IFO,App] Processing event: x 12095.0y  -23662.5	channel: 0	time: 0	position: 0
2024-02-18T16:14:41.280092 tid5452 [IFO,App] checking for sequencing
2024-02-18T16:14:41.280373 tid7720 [IFO,App] got event: x 12095.0y  -23662.5	channel: 1	time: 0	position: 0
2024-02-18T16:14:41.280551 tid5452 [IFO,App] Processing event: x 12095.0y  -23662.5	channel: 1	time: 0	position: 0
2024-02-18T16:14:41.280555 tid5452 [IFO,App] checking for sequencing
2024-02-18T16:14:41.280915 tid5452 [IFO,App] executing acquisition event
2024-02-18T16:14:41.286129 tid7720 [IFO,App] got event: x 12095.0y  -5662.5	channel: 0	time: 0	position: 0
2024-02-18T16:14:41.286143 tid7720 [IFO,App] acquisition aborted
```

nicost, 2024-02-18:
As far as I can tell that should only happen when the user aborts the acquisition (i.e., stops it somehow).  I have no idea what can trigger this.  Could you run this again, but then with debug logging enabled (Tools > Options)?  You can also use Help > Report a Problem to generate a debug level log.

I have been unable to reproduce this with the demo configuration.  It would be extremely useful if you can reproduce this behavior with the demo configuration, and post a protocol here.

ilyasdc, 2024-02-20:
> As far as I can tell that should only happen when the user aborts the acquisition (i.e., stops it somehow). I have no idea what can trigger this. Could you run this again, but then with debug [...]

Hello Nico,
Out of curiosity I tried the latest build (17 February) and it now acquires ! There's still a problem with autofocus not working as intended but I will open a new issue.

nicost, 2024-02-20:
Nice!  Thanks for that well described issue.  I hope to work on that soon.

PR #1893 (2024-02-17): Studio: AcqEngJ: fixed null pointer exception.
Not sure why this did not show up before, but skipFrames=0 (more or less the default) cause a divide by null exception.  Wondering if this also fixes #1891.
closed 2024-02-20 by nicost