micro-manager/micro-manager #1905: Request for new autofocus plugin

nicost, 2024-03-05:
While discussing code needed for lightsheet microscopes, we realized there is a need for a new aufocus plugin in Micro-Manager that has some of the capabilities of the diSPIM plugin in 1.4.  

Desired features are:
- It should be a superset of the Oughtafocus plugin, in addition it should:
- allow setting the z-stage to use for focusing (which can be different from the core-focus drive)
- Optionally run a Z-stack and interpret this after the fact rather than run a Brant optimizer.  In this mode, it should run a sequence if the hardware allows it (this could most easily be done using the acquisition engine), and display a focus quality curve if requested

Optional: the code could be structured such that it could accept a DataStore and provide a focus score per image in the store.  

Name?

fadero, 2024-03-05:
These features requests sound reasonable.

Unsure of a name. Some suggestions:

- Focus Stack
- Sharp Focus
- Focal Sweep

jondaniels, 2024-03-13:
I'm not picky as to the name.

Do you see this standing alongside OughtaFocus (and use the same scoring algorithms) or would it be augmenting OughtaFocus with some additional properties to choose whether to do the iterative algorithm or the stack + fit algorithm?

bls337, 2024-03-19:
I think the autofocus graph in the diSPIM plugin looks nice, it may be worth reusing some of the code. It has the code for curve-fitting in there too.

PR #1950 (2024-05-21): Adds ability to select the most in focus Z stack in multiple places.
- OughtaFocus plugin: adds option to run a complete Z stack and display the focus curve. Also adds the option to select a ZStage other than the default ZStage.
- FrameCombiner:  adds option to pick the sharpest frame (rather than combine features from multiple frames).
- ZProjector: adds option to select the sharpest frame.

Fixed bugs/confusion in Duplicator plugin and ImageProcessing library.

Closes #1905
closed 2024-05-21 by nicost