micro-manager/micro-manager #1948: Hardware communication differences with new acquisition engine?

dpshepherd, 2024-05-17:
Hi all,

We have a microscope that uses a ASI LX-4000 to control an XY stage, Z stage, and FW-1000 filter wheel. When we setup an MDA with multiple z positions and multiple channels, we encounter different behavior between the original acquisition engine and new experimental engine. It doesn't matter what order we acquire in, e.g. `[slice, channel]` or `[channel, slice]`

With the original acquisition, the acquisition completes fine. We can repeat it N times, do XY positions, etc.. and it all works well.

With the new experimental engine, we immediately get a serial error that the status of the one components connected to the LX-4000 cannot be read. For example, the position of the z stage cannot be read. Interestingly, the acquisition continues and then completes. The error also occurs if we run the same acquisition in pycromanager, which I believe uses the new experimental engine.

What can we do to help troubleshoot this? One guess I have is that the new engine is either sending too many commands to the LX-4000 or not waiting appropriately for a response before sending another command. 

Thanks!

marktsuchida, 2024-05-17:
It would be super helpful if you could collect a Problem Report (under the Help menu), which should log much of the relevant information (including the serial commands being sent). Best if you run a minimal acquisition that reproduces the issue, and also include a second report that uses the old acquisition engine for comparison.

dpshepherd, 2024-05-18:
Ok, here are the logs for two identical z stack with three channels, the first with the original engine and the second with the new engine. The original engine completes with no error, the new engine immediately pops up a serial port error with the LX-4000.
[new_acquisition_engine_error.txt](https://github.com/micro-manager/micro-manager/files/15365798/new_acquisition_engine_error.txt)
[original_acquisition_engine_noerror.txt](https://github.com/micro-manager/micro-manager/files/15365799/original_acquisition_engine_noerror.txt)

nicost, 2024-05-20:
Issue seems to be here:

```
2024-05-18T08:05:44.598181 tid11772 [dbg,Core] Did set camera BSIExpress exposure to 120.000 ms
2024-05-18T08:05:44.598227 tid11772 [dbg,Core] Config group LED: will apply preset F-Red
2024-05-18T08:05:44.598230 tid13356 [IFO,App] Updating GUI; config pad = true; from cache = true
2024-05-18T08:05:44.598252 tid11772 [dbg,Core:dev:Arduino-Switch] Will set property "Label" to "Red"
2024-05-18T08:05:44.598317 tid13356 [dbg,dev:COM5] SetCommand -> 1HW Z\r
2024-05-18T08:05:44.598322 tid11772 [dbg,Core:dev:Arduino-Switch] Did set property "Label" to "Red"
2024-05-18T08:05:44.598368 tid11772 [dbg,Core:dev:ASIFilterWheel] Will set property "State" to "2"
2024-05-18T08:05:44.598393 tid11772 [dbg,dev:COM5] SetCommand -> 3FMP 2\r
2024-05-18T08:05:45.098607 tid11772 [IFO,dev:COM5] TERM_TIMEOUT error occured!
2024-05-18T08:05:45.098726 tid11772 [ERR,Core] Error occurred in device COM5: Error in device "COM5": (Error message unavailable) (107)
2024-05-18T08:05:45.098784 tid11772 [ERR,Core:dev:ASIFilterWheel] Cannot set property "State" to "2" [ Error in device "ASIFilterWheel": Serial command failed.  Is the device connected to the serial port? (14) ]
2024-05-18T08:05:45.098828 tid11772 [dbg,Core:dev:ASIFilterWheel] Will set property "State" to "2"
2024-05-18T08:05:45.098858 tid11772 [dbg,dev:COM5] SetCommand -> 3FMP 2\r
2024-05-18T08:05:45.099568 tid13356 [IFO,dev:COM5] TERM_TIMEOUT error occured!
2024-05-18T08:05:45.099612 tid13356 [ERR,Core] Error occurred in device COM5: Error in device "COM5": (Error message unavailable) (107)
2024-05-18T08:05:45.099653 tid13356 [ERR,Core] Error occurred in device ZStage: Error in device "ZStage": Serial command failed.  Is the device connected to the serial port? (14)
2024-05-18T08:05:45.100108 tid13356 [IFO,App] Failed to get stage position
                                    [       ] java.lang.Exception: Error in device "ZStage": Serial command failed.  Is the device connected to the serial port? (14) in Thread[AWT-EventQueue-0,6,main]
```

I believe the new acquisition engine parallelizes talking to devices.  In this case that is a problem since the filter wheel and Z stage are behind the same COM port and the device does not tolerate multiple requests simultaneously.  

Is this the Tiger controller?  If so, you could try to make a configuration file with the Tiger device adapter, I would hope that handles this situation better.  

I see other instances in the log of "overlapping" commands and answers that get lost.  This should all be handles better if Micro-Manager understands these are all behind the same hub device.

dpshepherd, 2024-05-20:
Thanks! This is a LX-4000 controller, not a Tiger, so unfortunately that won't work.

We also see a similar issue when using an Arduino running the [micromanager firmware](https://micro-manager.org/Arduino) as the system shutter plus digital out control for LED power using the new acquisition engine. Doesn't happen with the original engine.

nicost, 2024-05-21:
Sorry, should have picked up on this.  It looks like the problem is caused by two different threads talking to the same LX-4000 at the same time.  One of these threads originates from the core.  It calls back to the UI through the "OnPropertiesChanged" callback.  The UI uses this as a signal to update its state, mainly from the System State Cache, but also by updating the stage positions.  Regretfully, the Z stage update request happens at the same time as the command to change the filterwheel position.  As a result (and as you suspected), the commands and answers to the LX-4000 controller get mixed up.  

I am not quite sure why this happens only with the new engine, but the main problem lies in multiple threads talking to the LX-4000.  Since it does not use a hub device, there is no mechanism that I can think of that guarantees these types of mix up can not happen.  The device adapters could be re-written to wait for the response after each command, but that would be extremely inelegant.

I could modify to callback code to not update the stage positions in this case.  However, I am not sure how that will affect the UI for others.

Curiously, the end of the log with the new engine shows many, many repeats of the following:
```
2024-05-18T08:08:44.610916 tid13356 [IFO,App] Updating GUI; config pad = true; from cache = true
2024-05-18T08:08:44.610948 tid13356 [dbg,dev:COM5] SetCommand -> 1HW Z\r
2024-05-18T08:08:44.616861 tid13356 [dbg,dev:COM5] GetAnswer <- :A 22812 \r\n\x03
2024-05-18T08:08:44.616927 tid13356 [dbg,dev:COM5] SetCommand -> 2HW X Y\r
2024-05-18T08:08:44.628841 tid13356 [dbg,dev:COM5] GetAnswer <- :A -16924.2 8611.9 \r\n\x03
2024-05-18T08:08:44.629076 tid13356 [IFO,App] Rebuilding config group table
2024-05-18T08:08:44.629275 tid13356 [IFO,App] Finished rebuilding config group table
2024-05-18T08:08:44.629842 tid13356 [IFO,App] Finished updating GUI
```
These are absent from the logs with the original engine.   All of these run on the UI thread (the EDT).  I wish I had an idea what triggers these callbacks, it seems something in the new engine, but I am not sure what (and these types of callbacks usually originate from the devices).

dpshepherd, 2024-05-21:
> I am not quite sure why this happens only with the new engine, but the main problem lies in multiple threads talking to the LX-4000. Since it does not use a hub device, there is no mechanism that I can think of that guarantees these types of mix up can not happen. 

Got it, sounds tricky to solve. 

Let me see if we can figure out a way to drive the camera and filter wheel in a hardware triggered loop for each Z plane to eliminate the filter wheel. We can definitely write a custom Arduino sketch or use an NI-DAQ for "DAQ as master", but this scope is in heavy use by people with less experience on the control code side.

As I mentioned, it also occasionally happens in the new acquisition engine with the Arduino as well, which I think is supposed to be a hub device? I will try to make a error log where we just repeatedly trigger an exposure and LED to see if I can grab that behavior.

> I wish I had an idea what triggers these callbacks, it seems something in the new engine, but I am not sure what (and these types of callbacks usually originate from the devices).

Perhaps they all come at the end after the user clicks "OK" on the blocking error window that pops up in micromanager and a bunch of delayed GUI update calls are made?

nicost, 2024-05-21:
Some more digging.  The callbacks originate from the PVCAM device adapter, which has the dangerous habit to call the OnPropertiesChanged callback whenever something changes.  In this case, the acquisition engine setting the exposure results in the OnPropertiesChanged callback prompting the GUI to query the Z stage and XY stage for their positions.  Interestingly, the old acquisition engine has the log: `Notification from MMCore ignored because acquisition is running!`.  

This log originates from `org.micromanager.events.internal.CoreEventCallback()` from the following lines:
```
     if (engine_ != null && engine_.isAcquisitionRunning()) {
         core_.logMessage("Notification from MMCore ignored because acquisition is running!", true);
```
So, with the new engine, it either is null or says that it is not running.  Turns out this code never becomes aware of the new engine, as it is created in MMStudio on line 394: `coreCallback_ = new CoreEventCallback(studio_, acqEngineClojure_);`.

Swapping out the hard coded Clojure engine for the acquisition manager should solve this.  

PR to this effect opened as #1952.

dpshepherd, 2024-05-21:
Thanks Nico! We'll give the fix a try once it merges.

nicost, 2024-05-23:
I believe this was in yesterday's nightly build (otherwise tonight).  Curious if this fixes the problem.  There clearly is still and underlying issue with the LX-4000, with multiple devices communicating through the same port at the same time, but the oppportunity for that happening should be greatly reduced now.

dpshepherd, 2024-06-06:
Hi all,

I spoke too soon - this error still pops in the micromanager GUI up when running a pycromanager acquisition. @henrypinkard - do you want to deal with it here or should I open an issue over at the pycromanager repository?

Thanks,
Doug

henrypinkard, 2024-06-07:
So if I understand correctly, the issue is that when running an acquisition from pycromanager, erroneous `onPropertyChanged` calls are causing the MM GUI to query things in device layer that they shouldn't be leading to some kind of multi-threading issue in the device? 

So the solution would seem to be to also check if Pycro-Manager acquisitions are running. @nicost do you see any issues with this change to make that happen: https://github.com/micro-manager/micro-manager/pull/1967

nicost, 2024-06-14:
I believe that works.  Did you test (I can not fully oversee the life cycle of "currentAcquisition_" and how the two acquisition types interact)?  Just quick tests with both in demo mode should show somethings.

henrypinkard, 2024-06-17:
It passes all the automated tests. @dpshepherd Let me know if you still see any problems

PR #1952 (2024-05-23): Studio: Make CoreEventCallback aware of the current acquisition engine.
Should fix issue #1948.

CoreEventCallBack was never told about the current acquisition engine.  My only worry is that the changed order of initialization in MMStudio may cause unforeseen problems.   Hope that Mark can review.

Closes #1948.
closed 2024-06-17 by henrypinkard