micro-manager/micro-manager #1970: Out of memory during MDA

ilyasdc, 2024-06-11:
MM version: 20240606 (20240607 seems to be the same)

Hello,

I wanted to try a more recent version of MM on our microscope (the HCS plugin works much better) but I was surprised to see that the acquisition failed quickly by showing a "system ran out of memory". The scenario to reproduce this issue is very simple:
- 2048x2048 images
- lots of positions (800 in our case)

Open ImageJ memory inspector and see the memory ramping up after which faills, 2000MB was allocated to ImageJ.
Under older micro-manager versions (I think we use the "stable" version on the website) this works (same 2000MB, memory never goes beyond 200-300MB)

Upon closing the MDA window all this memory gets released, so I guess it's not as leak but something tied to how the viewer works.

I attach a bug report, just in case but I don't think it's useful since how minimal the scenario to reproduce the error is.

[Problem Report.txt](https://github.com/user-attachments/files/15785953/Problem.Report.txt)

nicost, 2024-06-12:
Were you saving the data during this acquisition, and if so, which format did you select?  

Do you have the option (Tools > Options) "Save XY positions in separate Image Stack Files" checked?

ilyasdc, 2024-06-13:
> Were you saving the data during this acquisition, and if so, which format did you select?
> 

Yes, images are saved and not stored on the RAM. As I said the imageJ memory usage used to be minimal for this scenario (around 200MB-300MB checked from the memory usage panel of IJ) under previous versions (I can fetch the exact one if required). And  we use the standard tiff format ("Image stack file" in the interface)

> Do you have the option (Tools > Options) "Save XY positions in separate Image Stack Files" checked?

Yes.

The oom is easily reproducibe using the demo camera, storing images locally, it just suffices to set the image size to something large enough (2048x2048), for the default 512x512 it doesn't seem to trigger the issue.

marktsuchida, 2024-06-13:
> I wanted to try a more recent version of MM

It might help to know which version you were previously using successfully -- then we might be able to determine exactly which change caused the issue.

ilyasdc, 2024-06-13:
> > I wanted to try a more recent version of MM
> 
> It might help to know which version you were previously using successfully -- then we might be able to determine exactly which change caused the issue.

> > I wanted to try a more recent version of MM
> 
> It might help to know which version you were previously using successfully -- then we might be able to determine exactly which change caused the issue.

Hello,
I did a "bissection" and the problem manifests for the first time in build  >= 20240426  (to be clear: not in build < 20240426)

ilyasdc, 2024-08-13:
Hello @marktsuchida, @nicost, I tried the most recent build (20240801) and the memory still builds up like crazy. Can you reproduce the issue and do you need more help (e.g. additional tests) to determine the underlying cause ?

marktsuchida, 2024-08-13:
Thanks for the reminder, @ilyasdc.

The only PR merged on Apr 26 is #1935, but it could well be an interaction between that PR and the series earlier memory-related changes.

<details><summary>Details of the PRs between #1928 and #1935</summary>

I checked that we're not missing any changes, since the history is a little complicated. The following commits were merged consecutively in the order listed:

```text
Date merged PR    Merge commit                             Abbreviated PR title
----------- ----- ---------------------------------------- --------------------
2024-04-16  #1928 0de2d2c2095bb2789a5cb69519325d2afbc5b191 (unrelated)
2024-04-19  #1930 f22756bf5213a42cb3ec0591dc3423ccd0799338 fix leak
2024-04-22  #1927 3f2d21db762560e8fe59d3d84e4795ec11cddac0 reuse buffer (*1)
2024-04-23  #1931 4c2e78a7ad16764581a9d9e6fd48f4d9c1fb8494 "revert previous" (*2)
2024-04-23  #1932 c1c192f373de642551fed2781ac02546a89c9a81 revert #1927 (*1)
2024-04-24  #1934 a14e0ec166cbe247c1e7d5d6efc2fb08ca5d9741 redo reuse buffer
2024-04-26  #1935 186c8ff54ad901f1d2a70fab47a3785e499bd992 revert direct buffer
2024-05-02  #1938
```

Notes:
`*1`: #1932 exactly reverts #1927.
`*2`: #1931 does not merge any actual changes.

</details>

The actual changes that occurred are those of #1930, #1934, and #1935.

If the problem happens in nightly 2024-04-26 but not in nightly 2024-04-24, it might be that the symptoms don't show up as fast when direct byte buffers are used. This would not be surprising, because direct byte buffer memory would not show up in ImageJ's memory monitor (it still does consume memory on the computer).

So my guess for the place to look most carefully would be the changes of #1934 (since #1930 looks harmless).

@nicost (who wrote these PRs): do you have any immediate insights? Could it be that holding on to even a single buffer per position ends up filling up memory when there are many positions?

ilyasdc, 2024-08-14:
> Thanks for the reminder, @ilyasdc.
> 
> The only PR merged on Apr 26 is #1935, but it could well be an interaction between that PR and the series earlier memory-related changes.
> Details of the PRs between #1928 and #1935
> 
> The actual changes that occurred are those of #1930, #1934, and #1935.
> 
> If the problem happens in nightly 2024-04-26 but not in nightly 2024-04-24, it might be that the symptoms don't show up as fast when direct byte buffers are used. This would not be surprising, because direct byte buffer memory would not show up in ImageJ's memory monitor (it still does consume memory on the computer).
> 
You are correct. I tried again the 2024-04-24 build but I monitored the direct memory using the following code (ran in the script window): and while the imagej monitor stays reasonable, the direct memory shoots up so the problem must be more ancient.

> So my guess for the place to look most carefully would be the changes of #1934 (since #1930 looks harmless).
> 
> @nicost (who wrote these PRs): do you have any immediate insights? Could it be that holding on to even a single buffer per position ends up filling up memory when there are many positions?

I'm currently running tests but it seems that the "bug" occured after build 2024-04-06

nicost, 2024-08-14:
If this worked before with direct memory, then you should have enough RAM in your computer.  Can you simply increase the memory allocation to ImageJ?  Quick back of the envelope calculation shows you should be good with about 8GB (8000MB).  

I don't fully remember the details, but I believe this was exactly addressing the issue of memory leaks (or inefficient memory usage)  when acquiring images at many positions.

ilyasdc, 2024-08-14:
> If this worked before with direct memory, then you should have enough RAM in your computer. Can you simply increase the memory allocation to ImageJ? Quick back of the envelope calculation shows you [...]

Hello @nicost,  I guess this will work, indeed. But I don't understand why so much memory is now required when it used to be much more memory efficient. Maybe 800 positions is not enough for the memory leak/inefficient memory usage to show ? Actually I don't undertand how the change from #1934 leads to a buffer per position and not reusing the same for all positions, but I guess my java is too weak to see the subtlety here.

nicost, 2024-08-14:
Things slowly start to get back to me...  I believe that the changes were mainly prompted by long time-lapse positions at multiple positions.  The code used to keep Buffers in memory for each of the 4GB tiff files that it created, causing the computer to run out of memory.  The changes I made resulted in there only be one buffer per position.  While working on this, I discovered that 8-bit and RGB images were not handled correctly by the Direct ByteBuffer code, so I reverted to using JVM memory.  In your case, there should not really be a different in the amount or memory being used, however, instead of the memory being outside of the JVM, it is now inside.  All you need to do is change the memory allocation for the JVM.  

So no, this is not a memory leak, or inefficient memory usage; the code keeps one image per position in the JVM memory rather than in memory outside of the JVM.

ilyasdc, 2024-08-15:
Hello and thank you for your reply
> Things slowly start to get back to me... I believe that the changes were mainly prompted by long time-lapse positions at multiple positions. The code used to keep Buffers in memory for each of the 4GB tiff files that it created, causing the computer to run out of memory.

In that case the total memory footprint should be at least equal than it used to be and for long aquisition decrease, correct ?  (just in case, we do use the 1 file per position setting)

> The changes I made resulted in there only be one buffer per position. While working on this, I discovered that 8-bit and RGB images were not handled correctly by the Direct ByteBuffer code, so I [...]
Temporarily I agree that's the best solution.
> So no, this is not a memory leak, or inefficient memory usage; the code keeps one image per position in the JVM memory rather than in memory outside of the JVM.

What I notice is that the total memory used by micro-manager has shot up. As you mentioned it now scales with the number of positions but not the number of timepoints, that's good but that doesn't explain why it was much lower before since it was supposed to be at least as high as the number of positions.  

I also don't think we experienced any issue with out of memory issue for high number of positions + high number of timepoints despite running on weak machines (16GB of memory).

The main point I want to make is that one now needs to somehow precompute the maximum memory needed before running the acquisition, otherwise it will fail (at the first timepoint which is certainly easier to fix than if it failed midway)

I will try to generate more number to evaluate the impact of this change in  our case.

ilyasdc, 2024-08-16:
I didn't realise a `MultipageTiffReader` was used per position/tif file until i read more carefully the code.

I made a test with just opening the same dataset (800 postions) and looped through the position with the play button, I did that for versions 2024-04-23 and 2024-04-24 (before and after the buffer reuse #1934 ). The two plots of the total memory used, I hope this shows the total memory requirement have significantly changed. I assume the allocated buffers have a finite lifetime and are cleared out somehow ?

build 04/23
[image]
build 04/24
[image]

I a also stumbled on some piece of code in the `MultipageTiffwriter` which seems to handle reusing bytebuffers a bit differently https://github.com/nicost/micro-manager/blob/41dea9c101bc9981062219698fc0c4ecd2414b5b/mmstudio/src/main/java/org/micromanager/data/internal/multipagetiff/MultipageTiffWriter.java#L264-L338 maybe something along those lines could be used ?

marktsuchida, 2024-08-20:
It sounds like one way had problems with @nicost's use case and the other way has problems with @ilyasdc's use case.

Or at least I'm assuming there was a reason to recycle the buffer (which is a temporary buffer holding the pixels during a read, not the buffer of the returned `Image`; the latter is still allocated on every read), as opposed to just allocating a new one every time. Before #1934 (or the earlier, reverted version of it: #1927), the temporary buffer was just allocated on every read.

It is not clear from the description of #1927 how important the recycling part was (as opposed to the comparison between direct vs regular byte buffers). So if it's not critical, maybe we can revert #1934, but presumably there was a fairly strong motivation for #1934?

If the recycling _is_ important for performance (and it'd be helpful to know under which conditions), then perhaps we need something like a shared, size-limited pool of buffers for recycling, instead of each reader having its own recycled buffer.

ilyasdc, 2024-08-21:
> It sounds like one way had problems with @nicost's use case and the other way has problems with @ilyasdc's use case.
>
> Or at least I'm assuming there was a reason to recycle the buffer (which is a [...]

Hi @marktsuchida,

Agreed, and that's what you did for the `MultipageTiffwriter`, a shared pool of buffers shared between all the instances of the writer, correct ? https://github.com/nicost/micro-manager/commit/f34c2b9a9e6b3d7adde2cedc7576fd09d32bba3b

marktsuchida, 2024-08-22:
Yes, the same idea (ignoring the use of direct buffers). (Here's the link to the same commit in this repo: f34c2b9a9e6b3d7adde2cedc7576fd09d32bba3b.)

The hope would be that the same behavior of discarding recycled buffers when a differently sized one is requested will work well enough in the reader case, too. I think this is likely because I expect it to be uncommon to alternate between datasets of differently sized images at high rate.

And the same argument for a small max size (e.g., 3) probably applies to the reader as well, though this will need to be confirmed empirically.

nicost, 2024-10-29:
Shared pool of buffers for readers is now implemented.  @ilyasdc, can you test the latest nightly build to see if memory usage is better in your use case?

ilyasdc, 2024-10-30:
> Shared pool of buffers for readers is now implemented. @ilyasdc, can you test the latest nightly build to see if memory usage is better in your use case?

Hi @nicost, thank you for implementing this ! This seems to work well on linux. Total memory used by MM seems to be around 1.6GB for 900 2048x2048 images and doesn't increase much. I haven't tested on windows because the last nightly is `MMSetup_64bit_2.0.3_20241023.exe`.  I will close the issue.

closed 2024-10-30 by ilyasdc