micro-manager/micro-manager #1984: deskew convention for interpreting z-step_um and related metadata

jondaniels, 2024-06-21:
With the help of @fadero we confirmed that the convention of the deskew code in the on-the-fly image processor isn't what I was expecting.  Tagging @nicost and @henrypinkard due to their involvement, but at the root this is a question of which convention.

When collecting datasets that need to be deskewed, the orthogonal distance between images does not match the distance translated along the direction of movement.  **Should the "z-step_um" metadata value to be set to the orthogonal distance or the distance translated?**  We need to define a convention.

The orthogonal distance is the voxel size of the raw data and is the convention used by the original diSPIM plugin in MM1.4.

The distance translated in the real world makes a certain amount of sense and is the convention used by the deskew code presently.

Here is a graphical representation of my question depicting as-acquired images.

[image]

Related, **can we introduce a metadata field to encode the angle of motion relative to the optic axis?**  If the metadata field is not present or zero it can be assumed that the motion is along the optic axis which is the usual case with e.g. widefield microscope and a focus device.  But if the metadata field is specified then the deskew plugin can know how to operate on the data and maybe in the future some viewers would even be able to correctly render the raw data without needing a deskew.  (To be clear I'm only asking for a metadata field, not anything with the viewer yet.)

nicost, 2024-06-24:
Agreed on the need for angle metadata.  Where should those live?

Also, I am a bit partial to defaulting to the blue arrow as the default (but not strongly).  It is kind of similar to the fact that we record the distance the z stage travels in the metadata, not the optical displacement of the sample (for which we would need to know the refractive indices of the sample and the immersion medium).   It seems slightly better to record what the hardware does, and leave interpretation to higher levels of software.  Not a strong preference at all.

jondaniels, 2024-06-24:
> Agreed on the need for angle metadata. Where should those live?

Probably the metadata field should live next to the 'z-step_um'.  I'm not sure the best name, maybe 'z-angle_deg'?

>  It is kind of similar to the fact that we record the distance the z stage travels in the metadata, not the optical displacement of the sample

That's a reasonable reason to go with recording the actual distance translated, the blue arrow in the diagram.

The different convention used in MM1.4 will die with it and no need to change the MM2 deskew code.  We just need an interim work-around so that MM1.4 data can be deskewed with MM2, which may or may not be trivial.

fadero, 2024-06-24:
> Also, I am a bit partial to defaulting to the blue arrow as the default (but not strongly). 

I strongly agree with @nicost's preference here. I think that **z-step_um** should reflect hardware changes that are as agnostic as possible to sample (e.g. refractive index) and other hardware (e.g. O2/O3 tilt angle) parameters, to maximize abstraction. 

> Probably the metadata field should live next to the 'z-step_um'. I'm not sure the best name, maybe 'z-angle_deg'?

@jondaniels's suggestion of **z-angle_deg** makes the most sense to me. In a traditional system (e.g. spinning disk confocal with a z-piezo), the value of **z-angle_deg** should be close to 0º. Suggestion: this could be a property of any z-stage (e.g. focus drive, z-piezo, perfect focus) that defaults to 0º.

nicost, 2024-06-25:
> We just need an interim work-around so that MM1.4 data can be deskewed with MM2, which may or may not be trivial.

Would adding in a user-setable factor (defaulting to 1.0) be OK?  Something like: "Multiply z-step by `1.0`"?

> Suggestion: this could be a property of any z-stage (e.g. focus drive, z-piezo, perfect focus) that defaults to 0º.

I like that.  @marktsuchida, what do you think?

jondaniels, 2024-06-25:
>> We just need an interim work-around so that MM1.4 data can be deskewed with MM2, which may or may not be trivial.

> Would adding in a user-setable factor (defaulting to 1.0) be OK? Something like: "Multiply z-step by 1.0"?

Yes I think so.  In the deskew UI right?

marktsuchida, 2024-06-25:
> I like that. @marktsuchida, what do you think?

I don't think it's a problem to add a property that specifies some hardware attribute, but would you mind clarifying (for my understanding) what the definition of this angle is? Is it the tilt relative to the optical axis? If so, the name should contain the word "tilt", not "angle" (note that rotational stages are currently modeled as "Z" stages in MM, so angle could be confusing).

jondaniels, 2024-06-26:
@marktsuchida, it's the angle between the sample/stage motion and the optical axis during an image stack.   This angle is 0 degrees in a "usual" microscope, but non-zero in multiple important light sheet modalities.  Usually the stage moves along a cardinal direction in the lab frame but the light sheet is tilted.  The figure in the first comment in this issue represents the most common scenario.

Somehow "angle" is a more clear descriptor to me than "tilt" because the sample isn't usually tilted, but rather the plane captured on the camera is tilted in the lab frame.  But this is a weak preference.

fadero, 2024-06-27:
After lengthy discussions with @nicost, we agreed that it might be useful to define some optomechanical conventions. This isn't necessarily a mandate that we adopt these terms as device properties, but it will at least allow us to communicate unambiguously, instead of using imprecise language like "tilt."

I've defined three rotational transforms (pitch, yaw, and roll) and highlighted two examples that should help disambiguate a "rotational stage" (properly: roll stage) from the "tilt" (properly: yaw) that is common on single-objective light sheet microscopes. 

[image]

[image]

[image]

marktsuchida, 2024-06-28:
This is super helpful. (And "tilt" was not a good suggestion, I agree.)

- I think for completeness' sake the definition of the optical reference frame also needs to specify the _direction_ of the X axis. (That, together with specifying that the coordinate system is right-handed, will fix the Y axis direction.)

- `z-step_um` is generated by the acquisition engine/plugin, never in the devices or Core. But I think I agree with @nicost and @fadero that it should be a raw mechanical value, and any information needed to convert it to the spacing of imaged planes should be recorded in separate metadata field(s) -- just the principle that raw data should be preserved without loss of information when practical. Of course, light-sheet-specific acquisition code could add a new field like `z-orthogonal-step_um`.

About the `z-angle_deg` property being proposed:

- Now that I understand a little better (sorry, I was trying to quickly answer when I didn't have the time to throughly read the whole issue), I'm not convinced that it belongs in stages. Isn't this a property of the stage + (primary) objective combination, that will differ, say, for the left and right objectives in the case of a diSPIM-like configuration (especially in the case where the stage is the horizontal XY stage)? If so, I think it behaves more similarly to a pixel-size-config than to a device property.

- Would something like "skew angle" make sense as a name? Possibly "x-z skew angle" (and theoretically there could also be a "y-z")?

- There is also the question of whether to record the angle or its tangent, i.e., the x-z skew ratio (X shift per orthogonal (green) Z shift, although X shift per mechanical (blue) Z shift is also a possibility). My relative outsider (to light-sheet) gut feeling is that the ratio is easier to interpret than the angle, but I'm not sure.

- I also think it would be nice if the definition of this value is interpretable without knowing (or thinking through) the entire optical path. Each dataset already has an X (camera row, left to right) and Z direction (however the slice positions are recorded), and it would make a lot more sense if the skew parameter defines the relationship between these, rather than any consideration about the layout of the microscope. Unless I'm missing something, which I could be....

jondaniels, 2024-06-28:
Hi @marktsuchida ,

> Isn't this a property of the stage + (primary) objective combination, that will differ, say, for the left and right objectives in the case of a diSPIM-like configuration (especially in the case [...]

This seems correct to me.

> Would something like "skew angle" make sense as a name? Possibly "x-z skew angle"

I like that idea.

> (and theoretically there could also be a "y-z")?

Yes, in fact if the camera is rotated by 90° then the x-z skew becomes a y-z skew.  In the MM1.4 diSPIM plugin we record the angle and then have a checkbox in post-processing that determines whether the angle is interpreted as x-z or y-z skew.  (And another checkbox that inverts the direction, basically adding 180° to the specified angle.)  But I think we either need to have separate x-z and y-z values or else yet another metadata field to indicate the direction of the skew.

>   There is also the question of whether to record the angle or its tangent, i.e., the x-z skew ratio

I have no preference here, as long as we define things so if the skew/angle is 0° then the metadata value is 0 and this is the default case for a "normal microscope."

>   Each dataset already has an X (camera row, left to right) and Z direction (however the slice positions are recorded), and it would make a lot more sense if the skew parameter defines the relationship between these, rather than any consideration about the layout of the microscope.

I fully agree.

fadero, 2024-06-28:
> I think for completeness' sake the definition of the optical reference frame also needs to specify the direction of the X axis. (That, together with specifying that the coordinate system is right-handed, will fix the Y axis direction.)

Great catch! I've updated the figure to include the directionalities of the X- and Y-axes.

> I'm not convinced that it belongs in stages. Isn't this a property of the stage + (primary) objective combination, that will differ, say, for the left and right objectives in the case of a [...]

No strong opinion on whether this SOLS yaw is a property that lives in pixel-size-config or on a stage. My point is that the yaw is a _relationship_ between a translational stage (e.g. a lens-galvo mirror-lens combo, a scanning sample stage) and a camera. 

> Would something like "skew angle" make sense as a name? Possibly "x-z skew angle" (and theoretically there could also be a "y-z")?

In theory, yes there could be a camera-stage pitch. Camera-stage yaw is arguably more common in SOLS, but disambiguating between the different possible types of skew (if we agree to define "skew" as the general concept of "a translational stage does not translate the camera plane at an angle of 0º to the camera's normal vector) seems prudent.

> I have no preference here, as long as we define things so if the skew/angle is 0° then the metadata value is 0 and this is the default case for a "normal microscope."

I agree with @jondaniels here- a "skew" that has a 0º pitch and 0º yaw should be the default case.

nicost, 2024-06-28:
Talking with @fadero I believe we came to the conclusion that yaw as in SOLS (but possibly also in diSPIM) belongs to the camera.  This could be part of the pixel size configuration.  Currently, the pixel size affine transform defines the relationship between movement of the stage and the image on the camera.  It does seem logically to belong there, just not sure how to most cleanly (and in a backwards compatible way) to include there.  Ideas?

marktsuchida, 2024-06-28:
I don't think it is a problem to add one or two entries to pixel size presets, especially since the default is zero. We can make sure to _not_ write these to the config file unless they are non-zero, so that the new feature is completely invisible to code that doesn't know about it.

But it might be good to sketch out how this will actually look, for representative single- and dual-objective scopes. What are all the properties that would need to be included in the pixel size config in order for the skew value to be correctly switched for all the possible combinations of objectives/cameras and stages/galvos on a given system?

nicost, 2024-07-01:
I may be getting back to thinking about this in terms of relation of the camera to the XY stage (where @fadero thoughts about optical axis as the reference frame are extremely clarifying).  

The current pixelsize configurations contains the affine transform describing the 2D relation between XY stage and camera, which includes roll.  I assume that pitch and yaw could be described using a 3D affine transform, but I am afraid that will become very difficult to understand, and also difficult (though not impossible) to measure.  

So, what about adding pitch and yaw (in degrees, as defined above) to the pixel size configurations?

marktsuchida, 2024-07-01:
I think my only concern is that the definition of pitch and yaw is complicated and hard to understand (and, perhaps more importantly, it is hard to notice that you need to read up on the above definitions before interpreting naïvely), unless they are more simply defined in terms of the image's XYZ axes (which have no particular relationship to the optical table). Which is why I proposed the "x-z" and "y-z" terminology using the image coordinates. I also think that ratio is easier than degrees (especially to get the sign right). I'm willing to be persuaded otherwise on both points, but so far have not been :)

fadero, 2024-07-01:
> unless they are more simply defined in terms of the image's XYZ axes (which have no particular relationship to the optical table). Which is why I proposed the "x-z" and "y-z" terminology using the image coordinates.

You can indeed form a logically consistent set of descriptions with your proposed "camera reference frame!" I've attached a modification to the above "optical axis definitions," which I'll now call "camera definitions."

Some notes: we assume the _camera_ is fixed, and everything (stages) move relative to the camera. That means that, in the SOLS case, the light sheet is static, and the entire microscope (O1, sample, etc) moves relative to the camera plane. 

You're partially correct that now you need to describe the stages as more than just single-axis stages. You can do this with "pitch, yaw, and roll," or you can do it as you suggested: with ratios. The most concise way to do this would be to assign each axis ("x" sample stage, "y" sample stage, "z" sample stage, galvo mirror) a 3D unit vector (in the reference frame of the camera, of course) that fully describes the relationship of the stage's movement relative to the three axes of the camera. 

Practically, if you do choose to use the camera as your reference frame, then you need to add *two additional device properties* to _each_ axis of _every_ stage. In the case of SOLS, this results in eight additional device parameters.

Alternatively, if we adopt the aforementioned "optical axis of the first objective lens" convention, then you only need to add *two additional device properties* only to the _camera_. These parameters can be pitch and yaw (given that sample stage/camera roll is already calculated in the 2D pixel affine transform), or they can be a 3D unit vector. 

To summarize, there are two points I'm making here:

- it's harder, although not impossible, to use the camera as the absolute reference frame
- 3D unit vectors (ratios of axes) can be used as a substitute for assigning pitch/yaw/roll values to necessary components

@nicost suggested a virtual meeting to continue discussing. I am happy to organize that if @marktsuchida and @jondaniels want to join.

[image]

[image]

[image]

marktsuchida, 2024-07-02:
> Practically, if you do choose to use the camera as your reference frame, then you need to add _two additional device properties_ to _each_ axis of _every_ stage. In the case of SOLS, this results in eight additional device parameters.

I think there is a difference between the image's coordinate system (my suggestion) and a mechanical coordinate system described in the camera's frame of reference.

I'm suspecting that we might be (perhaps rightly) talking about two separate needs here (and/or I might be misunderstanding the scope of the problem). How many parameters are needed to correctly deskew a stack, from the point of view of the deskewing code? My thinking above was that those parameters could simply be recorded. And this (in my rather basic understanding, at least) does not require any encoding of the camera-objective-stage relationships, only the relationships between the axes of the resulting image stack (similarly to our affine transformations for the 2D case).

One thing I know I haven't considered (and I don't think has been stated above, though perhaps it was implied without my realizing) is deriving a common absolute coordinate system for position within the specimen, across different stepping mechanisms. This would allow mapping stacks acquired by stepping the stage vs galvo onto the same 3D space. If this is to be part of the goal, then presumably (at least in the general case) we need a transformation to be recorded for each positioning device axis (and forcing this into the pixel size config would probably not work well).

It would help to clarify the goal (though recording the simpler parameters for deskewing might still be useful even if we also (optionally?) support absolute coordinate mapping).

> @nicost suggested a virtual meeting to continue discussing. I am happy to organize that if @marktsuchida and @jondaniels want to join.

Absolutely!

jondaniels, 2024-07-06:
> Practically, if you do choose to use the camera as your reference frame, then you need to add two additional device properties to each axis of every stage. In the case of SOLS, this results in eight [...]

It seems cleanest to make this analogous to a pixel calibration.  E.g. next to the camera's affine transform the user would specify two numbers corresponding to pitch and yaw (colloquially the skew along x-z and y-z).  I agree with @marktsuchida that ratios will be less confusing than angles.

> How many parameters are needed to correctly deskew a stack, from the point of view of the deskewing code? 

I think just these two pitch/yaw parameters.

> ... deriving a common absolute coordinate system for position within the specimen, across different stepping mechanisms ...

I don't see a need for an absolute coordinate system sufficient to justify the extra effort/complexity of adding two parameters to every stage axis.

> @nicost suggested a virtual meeting to continue discussing. I am happy to organize that if @marktsuchida and @jondaniels want to join.

I would be happy to join but I will be hard pressed to join until Friday July 12.

To revisit an earlier point:

>>    We just need an interim work-around so that MM1.4 data can be deskewed with MM2, which may or may not be trivial.

> Would adding in a user-setable factor (defaulting to 1.0) be OK? Something like: "Multiply z-step by 1.0"?

Yes I think that plus some sort of checkbox for X vs. Y skew will be fine.  Or alternatively is there a way to add these new metadata fields post-acquisition?

jondaniels, 2024-08-15:
@fadero and @nicost and @marktsuchida : would it be worth having a virtual meeting to wrap this up and agree on an action plan?

fadero, 2024-08-16:
@jondaniels @marktsuchida @nicost sent you all a calendar invite for next week to chat virtually about this. Please feel free to move the calendar event to a different day/time.

fadero, 2024-09-03:
@jondaniels @marktsuchida @nicost Here's a draft of the first part of the dx/dz & dy/dz convention. I'm going to extend this convention to oblique plane microscopes tomorrow. Please review and let me know if you see any errors. 

[image]

fadero, 2024-09-05:
@marktsuchida @nicost @jondaniels I think I've finished my draft of the dx/dz & dy/dz convention. It includes normal cases, as well as the single objective light sheet case. I believe that I've made it general enough that we can define dy/dz and dx/dz from knowing the angle that the scan axis (e.g. galvo mirror, stage scan) makes with the camera's x-axis and y-axis. 

Unless there are any glaring errors, I think this is everything that @marktsuchida and @nicost need from me to get started on the new core properties and changes to the deskew plugin, respectively. 

[image]

[image]

marktsuchida, 2024-09-05:
> right-handed: the left thumb

Shouldn't a right-handed system have right index/middle/thumb pointing in +x/+y/+z?

fadero, 2024-09-05:
> Shouldn't a right-handed system have right index/middle/thumb pointing in +x/+y/+z?

It should have read "left-handed." I corrected the error, thanks for the catch

jondaniels, 2024-10-31:
@nicost and @marktsuchida any chance to work on this further?  From my notes we talked about:

1. Coming up with a diagram/examples to document the convention.  (I think @fadero did this.)
2. Adding dx/dz and dy/dz values to the Core, including saving in the config file and metadata.  For backward compatibility the values would be 0 unless specified.  
3. Adding a way to specify dx/dz and dy/dz in the GUI (maybe next to pixel size or maybe somewhere else).
4. Changing the deskew image processor to replace the user-specified angle parameter with dx/dz and dy/dz which can be pre-populated from from metadata (but should be user-editable if needed).

marktsuchida, 2024-10-31:
Yes, thanks for the nudge. I need to put out some fires this week but should be able to get to the Core part next week.

jondaniels, 2025-01-07:
Nudging again ;-) has there been any progress?  I am afraid to try my hand at edits to the Core or central GUI elements.

marktsuchida, 2025-02-19:
See also: #2059.

Remaining step is to document this on the [website](https://github.com/micro-manager/micro-manager.github.io), I think.

nicost, 2025-03-26:
Finally getting around to working on the deskew plugin.

Couple of questions:  we currently have the angle in degrees.  I believe that it is related to the new way of defining these relations as dx/dz expressed in degrees.  Is it OK to replace the angle with dx/dz (i.e. tan(degree))? 

Also, I need to figure out how to  implement deskewing dy/dz != 0 data.  That code lives in PycroManager (cpu) and Clij(GPU).  Does anyone have example data so that I can check things work as expected.  Is there a real world use case for dy/dz != 0.00?

fadero, 2025-03-27:
> Is it OK to replace the angle with dx/dz (i.e. tan(degree))?

Seems logical, but you may want to test edge cases (e.g. negative values)

> Also, I need to figure out how to implement deskewing dy/dz != 0 data

I think you are confused here. Normal SOLS data has dy/dz != 0 (e.g. https://x.com/tanfad/status/1694752624858570992?s=46). dx/dz is normally zero.

I thought about it some, and there's not really a good use case that requires dx/dz deskewing. Potentially if someone built their SOLS wrong and introduced some nonzero dx/dz along with dy/dz. But it's an edge case that's not worth solving, at least immediately.

closed 2025-03-27 by nicost

jondaniels, 2025-03-31:
@nicost wrote:

> Also, I need to figure out how to implement deskewing dy/dz != 0 data. That code lives in PycroManager (cpu) and Clij(GPU). Does anyone have example data so that I can check things work as expected. Is there a real world use case for dy/dz != 0.00?

This basically is a question of how you have the camera oriented.  If rotated by 90 degrees then you have dx/dz instead of dy/dz.

ASI has datasets that are skewed in both directions, so it would be great to have both options available in the deskew plugin.  In the short-term datasets could be rotated and then deskewed.  I agree that only one of dx/dz and dy/dz would be non-zero.

I will try to do some testing this week but it might take a bit longer.

PR #2083 (2025-03-27): Deskew plugin: Now uses dy/dz for angle when available.
Complicated ways to prefer the config file value but honor changes.

Closes #1984.  

When there are deficiencies in this approach, please open a new issue (mentioning #1984).