micro-manager/micro-manager #1986: MM2 currently does not support 64-bit RGB mode (16bit colour cameras)

Victaphanta, 2024-06-24:
Hi,
I have purchased a Tucsen colour cmos camera (Michrome 20mp, 16bit) for my stereo microscope with a motorized zdrive (Leica M205c) and I have been using the provided TUCam device adapter to integrate it into micromanager 2 (very recent nightly build). All seems to work fine at 8 bit, but MM2 fails to handle 16bit. Essentially, live freezes or if I try to take a snap I get the error “Missing pixel type”. In the core log I get this:

“Failed to snap image
[ ] java.lang.IllegalArgumentException: Missing pixel type in Thread[AWT-EventQueue-0,6,main]”

Looking at the Tuscsen device adapter code in VS all seems fine.

Interestingly, when I reverted back to MM 1.4 then all works fine, I am able to get live view and acquire images at 16bit at all resolution levels no problems. The only annoying thing is that the scale bar is misconfigured for some reason, but I can live with that as I can do this through imagej itself.

I posted this issue on image.sc and Nico had recommended I raise it here after confirming this issue exists.  See,
https://forum.image.sc/t/are-16bit-colour-cameras-supported-by-micromanager-2/98070

Please, is it possible to get back support >8bit colour cameras (in my case need support 64bit RGB).  

Many thanks for providing and maintaining MM, it is very much appreciated.

marktsuchida, 2024-06-28:
This is going to be difficult, unfortunately, mostly because the underlying ImageJ viewer (canvas) that MM uses to draw the image does not truly support 16-bit RGB images.

If I remember correctly, MM 1.4 was able to display such images by treating them as 3-channel images (that is, 3 separate planes for R, G, and B, instead of a single plane whose each pixel contains RGB), but outside of the display, this did not mesh well with the data representation and I think there were quite a few issues. And in any case it's not really correct to treat an RGB image as a 3-channel image in most uses of RGB images.

But that doesn't mean we need to give up, or course. The best way I can imagine this working would be to

- Add the capability to save and load 16-bit RGB TIFFs (which should be relatively straightforward), and
- Add white balancing support to the image viewer as a separate step before histogram computation. Make this reduce the image to 8-bit RGB before all the other steps of displaying, so that the existing viewer code can handle the image with minimal modification.

There are actually people in my own lab who would also benefit from this, so it would be nice to do -- but it's a fairly large amount of work so it's difficult to promise any timeline, I'm afraid.

MM 2.0 also has an incomplete implementation of histograms for 8-bit RGB (#1443, #1444), which I do intend to revisit.

Victaphanta, 2024-07-12:
Hi Mark, sorry for the delayed response, have been on leave with the kids for the school holidays. 

It would be great to get this capacity back and work within the MM2 environment as we have not had any issue with the images treated as 3-channel images.  But I am interested to know why it is not the correct way to treat RGB as 3 channels.

Regarding your second point above, I do not play around with the histogram as changes there do not manifest/ apply to acquisition.  They seem to only apply to the live images.  When I changes camera settings, such as gamma, sharpness, white balance etc, via device settings, those settings apply to both live view and acquisition.  This really did my head in for a while.  My procedure now is to leave the histogram alone (full range), and only use the device settings to adjust the image before acquisitions.   This works fine, but out of curiosity, why was it decided to unlink the histogram settings from the device setting.  

On a similar note, my usage is for deriving z-stacks for montaging work - museum specimens in particular.  In the past we have been using Leica's LAS software modules, and in that system the live view (which was linked to the device settings) could be set to a lower resolution, while the acquisition could be set to a higher resolution.  In short, this allowed live viewing to be faster and more responsive, but the actual acquisition resolution be at the highest or most desired level.  Is there a way a similar separation can be achieved in MM.  I have not been able to crack that one.     

Many thank for your help.