micro-manager/micro-manager #2002: [Feature request] adjust custom HCS templates sites sizes

ilyasdc, 2024-08-13:
Hello,

When creating a custom HCS template, it is not possible to adjust the extent of the sites, on the contrary it seems that this property, called `_wellSpacing` is retrieved from the previously selected template, the values are hardcoded and this renders HCS harder since the displayed dimensions don't correspond to the real ones, unless the plate dimensions cosely the one we are using. In our case we used 24 well-plate  which are larger than the 24 well-plate templae (ours: 16200 um large, hardcoded: 14000)  https://github.com/micro-manager/micro-manager/blob/41dea9c101bc9981062219698fc0c4ecd2414b5b/plugins/HCS/src/main/java/org/micromanager/hcs/SBSPlate.java#L135

I think the modifications aren't that extensive since it's just a matter of adding a field, so I could give it a go even though my java knowledge is rather limited.

nicost, 2024-08-14:
I am not quite sure I understand the issue.  The code to write (and read) custom layouts is copied below and looks correct to me. There is no variable `_wellSpacing`, but there are variables `wellSpacingX_` and `wellSpacingY_` and those are set correctly.  
Do you have a specific example including the customSetting file that you created?

Also, what is your definition of the "extent of the sites"?

```
   public String serialize() throws HCSException {
      return serialize(numRows_, numColumns_, wellSpacingX_, wellSpacingY_,
            sizeXUm_, sizeYUm_, id_, description_, firstWellX_, firstWellY_,
            circular_);
   }

   public static String serialize(int numRows, int numColumns,
                                  double wellSpacingX, double wellSpacingY, double sizeXUm,
                                  double sizeYUm, String id, String description, double firstWellX,
                                  double firstWellY, boolean circular) throws HCSException {
      JSONObject plate = new JSONObject();
      try {
         plate.put(ROWS, numRows);
         plate.put(COLS, numColumns);
         plate.put(WELL_SPACING_X, wellSpacingX);
         plate.put(WELL_SPACING_Y, wellSpacingY);
         plate.put(PLATE_SIZE_X, sizeXUm);
         plate.put(PLATE_SIZE_Y, sizeYUm);
         plate.put(ID, id);
         plate.put(DESCRIPTION, description);
         plate.put(FIRST_WELL_X, firstWellX);
         plate.put(FIRST_WELL_Y, firstWellY);
         plate.put(CIRCULAR, circular);
         return plate.toString(3);
      } catch (JSONException e) {
         throw new HCSException(e);
      }
   }

   public void restore(String ser) throws HCSException {
      JSONObject plate;
      try {
         plate = new JSONObject(ser);
         numRows_ = plate.getInt(ROWS);
         numColumns_ = plate.getInt(COLS);
         wellSpacingX_ = plate.getInt(WELL_SPACING_X);
         wellSpacingY_ = plate.getInt(WELL_SPACING_Y);
         sizeXUm_ = plate.getInt(PLATE_SIZE_X);
         sizeYUm_ = plate.getInt(PLATE_SIZE_Y);
         id_ = plate.getString(ID);
         description_ = plate.getString(DESCRIPTION);
         firstWellX_ = plate.getDouble(FIRST_WELL_X);
         firstWellY_ = plate.getDouble(FIRST_WELL_Y);
         circular_ = plate.getBoolean(CIRCULAR);
      } catch (JSONException e) {
         throw new HCSException(e);
      }
   }
```

ilyasdc, 2024-08-14:
Hello @nicost, sorry for the confusion, I was quite tired yesterday. The variable that I'm looking for is `wellSize` ( I correctly indicated the code location but not the variable name ...), this one is not saved/retrieved, correct ?

nicost, 2024-08-14:
Ah!  `wellSizeX` and `wellSizeY`.  Yes, those are missing.  Added in PR #2003.

PR #2003 (2024-08-14): HCS plugin: Adds wellSizeX ad wellSizeY to custom Plate editor.
Closes #2002
closed 2024-08-14 by nicost