micro-manager/micro-manager #2004: Setup mm2 with Nikon Eclipse TiE

shay82, 2024-08-18:
I'm attempting to connect my Nikon Eclipse TiE microscope to control the Z-Axis (focus) using Micro-Manager 2. 
Although I have installed the microscope driver and can control it through the Ti Control app, I am unable to connect it to Micro-Manager.

_Device drivers in windows device manager:_
[image]

_MicroManager_ device configuration wizard
[image]

_Questions:_

1. Which device should I select in the Device Configuration Wizard for the Nikon Eclipse TiE microscope: the first or the second option?

2. The Nikon TE 2000 in the Device Manager, which seems most relevant, requires port configuration. However, the microscope is connected via USB 3.0, and my computer doesn't have a COM port. How can I use the USB port for this connection? Is it possible to utilize a USB serial port with a virtual COM port, and if so, how can I set this up?

3. I have a microscope camera connected via USB, and I am capturing frames using OpenCV (cv2). Is there a way to map the camera to a USB serial port? Is this dependent on the camera driver, or is there a general solution available

marktsuchida, 2024-08-19:
`NikonTI` is the right one for the Ti-E. TE-2000 is an older model that uses a different protocol.

What is the error message (or other symptom) you get when trying to "Add" the `TIScope`?

I do not believe there is a way to map the camera to a USB serial port (and in any case, Micro-Manager does not directly use serial ports for cameras). If you want to talk to the camera in Micro-Manager, you will need to use one of the Micro-Manager device adapters (there is one based on OpenCV, though it is a little antiquated so I don't know if it will work the same as what you are doing currently).

shay82, 2024-08-20:
> `NikonTI` is the right one for the Ti-E. TE-2000 is an older model that uses a different protocol.
>
> What is the error message (or other symptom) you get when trying to "Add" the `TIScope`?
>
> I do not [...]

Thank you @marktsuchida for your response.

It seems that after a few configuration modifications, the  Ti-E works as expected.
Regarding the camera - I have check with my provider if they have specific camera to use in mm2 or should I use cv2.

closed 2024-08-20 by shay82