micro-manager/micro-manager #2006: Question: Pixel intensities are very high?

snguyen49, 2024-08-20:
Hi,

Originally, I was using MOSAIC for a TUCSEN camera, but it has not been working for me so I switched to Micromanager. However, when I used MOSAIC, the pixel intensities of my images ranged around 600-800. With micromanager, my intensities are in the 8000s. Is there a way to change this? All of the imaging parameters and camera are set the same (gain, exposure time, etc.), so I was wondering what could be causing this.

Many Thanks,
Sydney Nguyen

marktsuchida, 2024-08-22:
You didn't mention the illumination but I'm assuming it's at the same level. (If the software is controlling the illumination it would be useful to rule out the possibility of the illumination turning on and off at different timings.)

Could it be a different bit depth setting of the camera (say, 10 vs 14 bit)?

If not, this might be a good question for TUCSEN (who also maintain the Micro-Manager device adapter for their cameras).

Some more questions that may help getting at the cause:
- Do the Micro-Manager images appear to have discrete values (i.e., do all the pixel values fall on non-consecutive integers, leaving gaps with no pixels)?. You might see this by enlarging the histogram and seeing if it is solidly filled.
- Do the background (dark) pixel values (mean and spread) look similar?

nicost, 2024-11-12:
Closing, as this seems to involve device settings rather than software issue (please re-open if that is not the case).  For device settings, please open an issue on https://image.sc.

closed 2024-11-12 by nicost