micro-manager/micro-manager #2010: Building on Unix

johnfmarkham, 2024-09-20:
I'm following the instructions to build MM on linux [here](https://github.com/micro-manager/micro-manager/blob/bcfb15d634dafe9f58bcaa3d6cb31c4bcc1e629a/doc/how-to-build.md) and the following command times out:

svn checkout https://valelab4.ucsf.edu/svn/3rdpartypublic/classext

I think this contains iconloader.jar so that when I try and build MM I am missing  com.bulenkov.iconloader.IconLoader. iconloader.jar is available [elsewhere](https://github.com/bulenkov/iconloader). Where in the MM directory tree  should I put it so that it is found?

marktsuchida, 2024-09-20:
If you have `somewhere/micro-manager`, you should place it in `somewhere/3rdpartypublic/classext/iconloader.jar`. Sorry for the trouble while we fix this.

nicost, 2024-11-12:
Subversion repo is back online (for the time being).

closed 2024-11-12 by nicost