micro-manager/micro-manager #2037: Option to hide the System config group

marktsuchida, 2024-12-17:
Some users say that it is confusing or accident-prone.
Probably could go in Tools > Options.

nicost, 2024-12-17:
From the main Window?  Yikes, that gets ugly.  I have not heard this request before.

marktsuchida, 2024-12-20:
There may be other ways to address the actual issue, which is that the config group table is in relatively small font, and for some people it's apparently easy to accidentally click the wrong row (and we can't rename `System` to something like `ZZSystem` to get it out of the way). On a complex-enough configuration, this can reset a whole bunch of things (the system has a lot of electronics/DAQ settings that are not obvious) and it can be tedious to recover the desired state.

I do agree that we should be careful about something like this, because there are clearly tradeoffs: hiding things is always confusing, too. Especially when you think about what happens when you edit and save the config file.

Maybe I should rather look for a local workaround, such as using a startup script instead of the System-Startup preset. Not ideal because it won't travel with the config file, but might cause fewer issues in the world. Or maybe design our config groups better.

closed 2024-12-20 by marktsuchida