micro-manager/micro-manager #2038: Request to add desired Z step size to the MDA window

nicost, 2024-12-17:
The optimal Z-step is a matter of debate, and bet let to the system integrator to decide.  The question then becomes where to store this data.  It seems logical to add this to the pixel size configuration.  It would be an editable value, to be set in the configuration.  It could default to 0.0 (indicating that no desired Z step size is set).  

If others are OK with this, it could be added to the work to be done in

marktsuchida, 2025-02-19:
See also: #2059.

Remaining step is to document this on the [website](https://github.com/micro-manager/micro-manager.github.io), I think.

nicost, 2025-02-20:
Added documentation: https://github.com/micro-manager/micro-manager.github.io/pull/165

closed 2025-02-20 by nicost