micro-manager/micro-manager #2040: Micromanager refuses to build after latest update

mz-celldom, 2024-12-18:
[Latest commit ](https://github.com/micro-manager/micro-manager/blame/946bcab1a1754c70d247a2806083c4cfd84d6c9d/plugins/ASIPLogic/src/main/java/com/asiimaging/plogic/PLogicControlFrame.java#L11)

Building micro manager according to the instructions on Linux yields: 
```ASIPLogic.compile:
 [mm-javac] Compiling 34 source files to /home/XXX/Distro/micro-manager/plugins/build/ASIPLogic
 [mm-javac] warning: No processor claimed any of these annotations: java.lang.FunctionalInterface,org.scijava.plugin.Plugin
 [mm-javac] /home/XXX/Distro/micro-manager/plugins/ASIPLogic/src/main/java/com/asiimaging/plogic/PLogicControlFrame.java:11: error: cannot find symbol
 [mm-javac] import com.asiimaging.plogic.ui.asigui.ProgressBar;
 [mm-javac]                                       ^
 [mm-javac]   symbol:   class ProgressBar
 [mm-javac]   location: package com.asiimaging.plogic.ui.asigui
 [mm-javac] /home/XXX/Distro/micro-manager/plugins/ASIPLogic/src/main/java/com/asiimaging/plogic/PLogicControlFrame.java:34: error: cannot find symbol
 [mm-javac]    private final ProgressBar progressBar_;
 [mm-javac]                  ^
 [mm-javac]   symbol:   class ProgressBar
 [mm-javac]   location: class PLogicControlFrame
 [mm-javac] /home/XXX/Distro/micro-manager/plugins/ASIPLogic/src/main/java/com/asiimaging/plogic/PLogicControlFrame.java:155: error: cannot find symbol
 [mm-javac]    public ProgressBar getProgressBar() {
 [mm-javac]           ^
 [mm-javac]   symbol:   class ProgressBar
 [mm-javac]   location: class PLogicControlFrame
 [mm-javac] 3 errors
 [mm-javac] 1 warning```

nicost, 2024-12-18:
Pinging @bls337 and @jondaniels.  Looks like there is a file/class (ProgressBase) missing.

bls337, 2024-12-18:
@nicost @mz-celldom Sorry about that, I just submitted a pull request #2041

nicost, 2024-12-19:
Closing as this should work now.  Please re-open if it does not.

closed 2024-12-19 by nicost