micro-manager/micro-manager #2055: BUild on Windows broken

nicost, 2025-01-27:
```
ant build-java

[mm-javac] C:\Users\nstuurman\projects\micro-manager\mmstudio\src\main\java\org\micromanager\internal\MMStudio.java:45: error: cannot find symbol
 [mm-javac] import mmcorej.CMMCore;
 [mm-javac]               ^
 [mm-javac]   symbol:   class CMMCore
 [mm-javac]   location: package mmcorej
 [mm-javac] C:\Users\nstuurman\projects\micro-manager\mmstudio\src\main\java\org\micromanager\internal\MMStudio.java:46: error: cannot find symbol
 [mm-javac] import mmcorej.MMCoreJ;
```

Looks like MMCoreJ_wrap.dll is not found.  I even copied the version from the installer to `build/Java` but that did not help.  Where does ant expect to find MMCoreJ_wrap (or is something else missing?

marktsuchida, 2025-01-29:
`ant build-java` requires `ant build-cpp` to have successfully completed (but I get a different error when that's not the case).

I'm pretty sure it also works if you build just MMCoreJ_wrap in Visual Studio. If that's not enough, maybe also run `ant jar` inside `mmCoreAndDevices/MMCoreJ_wrap/`.

(FWIW, I think that error indicates that the CMMCore Java class missing, not the DLL missing.)

nicost, 2025-02-03:
Thanks!  Building MMCoreJ_wrap in Visual Studio, then running `ant jar` inside mmCoreAndDevices/MMCoreJ_wrap/  made it work.

closed 2025-02-03 by nicost