micro-manager/micro-manager #2075: pgFocus plugin fails to load in recent nightly builds

kmdouglass, 2025-03-06:
I am resurrecting an old custom-built Microscope setup. In the process, I have upgraded Micro-Manager to a recent nightly version and everything works except for the pgFocus plugin.

I realize that I might be the only person who uses this, so I am opening this issue to track the progress of my investigation.

## Steps to reproduce

1. Download and install a recent nightly version. I am using 20250228.
2. Create a new hardware configuration and add the pgFocus hub adapter and Stabilization component.
3. Navigate to `Plugins` > `Device Control` > `pgFocus`
4. Observe the following error:

> Error: pgFocus::getStringValue() Error reading values from pgFocus: Calibration curve

[image]

## Additional details

- The plugin works fine on the old version of Micro-Manager that the setup last used: 20180108
- The commit history for [PGFocusFrame.java](https://github.com/micro-manager/micro-manager/commits/main/plugins/pgFocus/src/main/java/edu/umassmed/pgfocus/PGFocusFrame.java) shows only one change between 20180108 and now: https://github.com/micro-manager/micro-manager/commit/b9ef0613c6bd85212541bc8bed9104c5baed1ca7

@kbellve are you still active on this project?

kmdouglass, 2025-03-06:
Looking at the code, it seems like the error is triggered here in a method called `updateValues`: https://github.com/micro-manager/micro-manager/blob/main/plugins/pgFocus/src/main/java/edu/umassmed/pgfocus/PGFocusFrame.java#L418

The line at the above link calls a helper function called `getStringProperty`:

```java
   String getStringProperty(String property) {
      try {

         String val;

         val = core_.getProperty(pgFocus_, property);

         return (val);

      } catch (Exception ex) {
         ReportingUtils.showError(
               "pgFocus::getStringValue() Error reading values from pgFocus: " + property);
      }

      return (null);
   }
```

Likely the `getProperty` of the core method is failing. Unfortunately the exception information from the core is dropped in the error handling.

kmdouglass, 2025-03-06:
I recompiled the plugin to log the exception that's being hidden:

> 2025-03-06T17:07:07.544392 tid1916 [IFO,App] java.lang.Exception: Cannot get value of property "Calibration curve" [ Error in device "pgFocus-Stabilization": Invalid property name encountered: Calibration curve (2) ]

kmdouglass, 2025-03-07:
I checked the device adapter and the property name is actually `Calibration Curve` (note the capital 'C' in 'Curve'). The plugin is trying to get the property `Calibration curve` (with a lowercase 'c'). I'll try recompiling either the plugin or the device adapter later today to see if this fixes it.

@marktsuchida @nicost Was the MMCore previously case insensitive when getting properties?

marktsuchida, 2025-03-10:
@kmdouglass I don't think it was ever case insensitive -- certainly not in the last decade. Maybe a schrödinbug?

closed 2025-03-10 by marktsuchida