micro-manager/micro-manager #2080: Issue with µManager and recent Matlab Versions

KonstantinNeuhaus, 2025-03-18:
I ran into some issue with connecting Matlab and µManager in recent Matlab Versions (tested with Matlab2022b and Matlab2024b).
If I follow the installation guide in the StartMMStudio script that comes with µManager, Matlab will not start after adding all the java class files to the javaclasspath.txt file.

However, if I remove the guava-17.0.jar from the classpath, Matlab work again and I can use µManager in Matlab.
Does anyone know if any issues can arise from this?
So far I used this solution for quite some time and did not run into any issues.

If no one sees a problem with not having the guava jar, I would like to contribute an updated Matlab installation guide and a modified StartMMStudio script.

Thank you for any input on this.

marktsuchida, 2025-03-18:
It sounds reasonable to me. If recent Matlab versions ship their own Guava (rather than the ancient google-collect, which they had at the time `StartMMStudio.m` was created), then it makes sense that there would be a conflict. In that case, excluding the Guava shipped with MM makes sense (especially if Matlab ships a newer Guava).

MMStudio would be very broken without a functioning Guava, so if it appears to work, it's probably going to be okay.

Thanks for looking into this.

KonstantinNeuhaus, 2025-03-19:
Actually, now that I wrote everything down again to update the Matlab Guide, I tested an adapted StartMMStudio.m script.
I then realized that it's not only the Guava.jar itself that caused the problem but only if its set with <before> argument.
So if Guava.jar is included as:
"<before>
C:\Micro-Manager-2.0\plugins\Micro-Manager\guava-17.0.jar"
to the Javaclasspath, Matlab will not start but if I include it without the <before> Matlab starts fine.

Would you still recommend to just exclude the guava.jar or to add it without the <before> statement.

marktsuchida, 2025-03-19:
Do you mean that the JAR file does not need to be deleted, but the entry in javaclasspath needs to be omitted? If so that sounds reasonable to me.

KonstantinNeuhaus, 2025-03-19:
Sorry, I just noticed that the command was interpreted as an actual command rather than displayed as text. 
I will try to clarify my question:
In the current StartMMStudio script, the Guava.jar is included in a way, that it is the first .jar file Matlab adds.
This is achieved with the <before command. 
During testing, I removed both the <before command and the Guava.jar file, and everything worked fine. To verify if the Guava.jar file was indeed causing MATLAB to fail to start, I tested further. I discovered that simply removing the <before command (while keeping Guava.jar in the classpath) also resolves the issue. It seems that the problem only arises when Guava.jar is placed at the beginning of the classpath using <before.

My question now is: Should we exclude Guava.jar entirely from MATLAB's javaclasspath, since it seems to work without it? Or would it be safer to just remove the <before command and keep the Guava.jar file in the classpath?

marktsuchida, 2025-03-19:
Ah, I see. I think that means that the script would work if all special handling of Guava (and google-collect) were to be removed.

Technically there could be problems with two different versions of the same library being on the class path, but that is not specific to Guava -- there may well be other Jars shipped by both Matlab and MM and we don't do anything special with them. I don't think there is a perfect solution to this that can be achieved with reasonable effort; we can try to deal with issues if they arise.

I guess we don't know which Matlab version replaced google-collect with guava. If you want to try to support all versions since R2013a, `AppendMMClasspath` could branch based on whether `google-collect.jar` was seen in javaclasspath: if yes, use the current code that special-cases Guava, and if no, just list all the jars, including Guava, with no `<before>`.

However, I suspect that Matlab has been shipping Guava for a while by now, so I think it is also fine to simply remove all the special handling of guava/google-collect from the script, treating it just like all the other jars.

KonstantinNeuhaus, 2025-03-20:
Yes, the StartMMStudio script still works with minimal changes.
The following addition to the AppendMMClasspath function is sufficient and would make it adaptable for older and recent Matlab versions.
```
   release = sscanf(version('-release'),'%d');
   if release > 2022
       fprintf(fid, '%s\n', guavaJar);
   else
       fprintf(fid, '<before>\n');
       fprintf(fid, '%s\n', guavaJar);
   end
```

I agree that it would be a more elegant solution to check the existence of the google-collect.jar in Matlab, but this might not be so easy. I checked a few Matlab Installations (2017b, 2018b, 2019b, 2020b,2021b, 2023b, 2024b) and all have a google-colllect.jar. 
I investigated this option and it seems like Matlab ships the up-to-date guava libraries but still calls them google-collect.jar.
I tried my modified StartMMStudio script under Matlab2021b, Matlab2022b and Matlab2024b and it works well on all of these versions.
I am also quite certain that the issue is only appearing in Matlab releases from 2022 on as their is a large jump in the google-collect/guava version that matlab ships:
2017b: 15.0.0
2018b: 16.0.1
2019b: 16.0.1
2020b: 16.0.1
2021b: 16.0.1
2022b: 30.1
2023b: 32.1.1
2024b: 32.1.1

I will create a pull request with the changes I described above.

KonstantinNeuhaus, 2025-03-27:
As this was my first proper pull request, I hope everything was correct.
Let me know if you need further input.
For the updated documentation, I am just waiting for a new device computer to confirm everything works just the same in Windows11 and with a fresh installation of everything.

marktsuchida, 2025-03-28:
Appreciate you going to the trouble of looking up the versions shipped with Matlab!

closed 2025-03-28 by marktsuchida