micro-manager/micro-manager #2085: Inspector windows is not opening in Micro-Manager

AndriiPidlisnyi, 2025-04-01:
Hi everyone!

First of all, I want to give a big thank you to everyone who has built and supported Micro-Manager.

Could you help me solve the following issue?
I installed the Micro-Manager on Ubuntu by following the guide here: https://github.com/micro-manager/micro-manager/blob/8b20e46a816586e13dac3b08d0b7db2b8d73db99/doc/how-to-build.md

For some reason, the inspector window is not opening when I capture a photo, and I see the following error in the console during program execution:

```
2025-03-04T09:41:06.066549 tid139960456754944 [IFO,App] Thread 15 (AWT-EventQueue-0) terminated with uncaught exception
2025-03-04T09:41:06.066593 tid139960456754944 [IFO,App] java.lang.NullPointerException
2025-03-04T09:41:06.066793 tid139960456754944 [IFO,App]   at org.micromanager.display.inspector.internal.panels.sharpnessinspector.ui.SharpnessInspectorPanel.addPropertyChangeListener(SharpnessInspectorPanel.java:190)
2025-03-04T09:41:06.066800 tid139960456754944 [IFO,App]   at javax.swing.plaf.synth.SynthPanelUI.installListeners(SynthPanelUI.java:83)
2025-03-04T09:41:06.066804 tid139960456754944 [IFO,App]   at javax.swing.plaf.synth.SynthPanelUI.installUI(SynthPanelUI.java:63)
2025-03-04T09:41:06.066808 tid139960456754944 [IFO,App]   at javax.swing.JComponent.setUI(JComponent.java:666)
2025-03-04T09:41:06.066812 tid139960456754944 [IFO,App]   at javax.swing.JPanel.setUI(JPanel.java:153)
2025-03-04T09:41:06.066816 tid139960456754944 [IFO,App]   at javax.swing.JPanel.updateUI(JPanel.java:126)
2025-03-04T09:41:06.066820 tid139960456754944 [IFO,App]   at javax.swing.JPanel.<init>(JPanel.java:86)
2025-03-04T09:41:06.066823 tid139960456754944 [IFO,App]   at javax.swing.JPanel.<init>(JPanel.java:95)
2025-03-04T09:41:06.066828 tid139960456754944 [IFO,App]   at org.micromanager.display.inspector.internal.panels.sharpnessinspector.ui.SharpnessInspectorPanel.<init>(SharpnessInspectorPanel.java:119)
2025-03-04T09:41:06.066832 tid139960456754944 [IFO,App]   at org.micromanager.display.inspector.internal.panels.sharpnessinspector.SharpnessInspectorController.<init>(SharpnessInspectorController.java:61)
2025-03-04T09:41:06.066837 tid139960456754944 [IFO,App]   at org.micromanager.display.inspector.internal.panels.sharpnessinspector.SharpnessInspectorController.create(SharpnessInspectorController.java:82)
2025-03-04T09:41:06.066842 tid139960456754944 [IFO,App]   at org.micromanager.display.inspector.internal.panels.sharpnessinspector.SharpnessInspectorPlugin.createPanelController(SharpnessInspectorPlugin.java:48)
2025-03-04T09:41:06.066846 tid139960456754944 [IFO,App]   at org.micromanager.display.inspector.internal.InspectorController.makeUI(InspectorController.java:204)
2025-03-04T09:41:06.066850 tid139960456754944 [IFO,App]   at org.micromanager.display.inspector.internal.InspectorController.create(InspectorController.java:129)
2025-03-04T09:41:06.066855 tid139960456754944 [IFO,App]   at org.micromanager.display.internal.DefaultDisplayManager.createInspectorForDataViewer(DefaultDisplayManager.java:281)
2025-03-04T09:41:06.066860 tid139960456754944 [IFO,App]   at org.micromanager.display.internal.gearmenu.GearButton.lambda$new$0(GearButton.java:48)
2025-03-04T09:41:06.066864 tid139960456754944 [IFO,App]   at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
2025-03-04T09:41:06.066868 tid139960456754944 [IFO,App]   at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
2025-03-04T09:41:06.066872 tid139960456754944 [IFO,App]   at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
2025-03-04T09:41:06.066876 tid139960456754944 [IFO,App]   at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
2025-03-04T09:41:06.066880 tid139960456754944 [IFO,App]   at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
2025-03-04T09:41:06.066884 tid139960456754944 [IFO,App]   at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:882)
2025-03-04T09:41:06.066888 tid139960456754944 [IFO,App]   at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:926)
2025-03-04T09:41:06.066893 tid139960456754944 [IFO,App]   at java.awt.Component.processMouseEvent(Component.java:6539)
2025-03-04T09:41:06.066897 tid139960456754944 [IFO,App]   at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
2025-03-04T09:41:06.066901 tid139960456754944 [IFO,App]   at java.awt.Component.processEvent(Component.java:6304)
2025-03-04T09:41:06.066904 tid139960456754944 [IFO,App]   at java.awt.Container.processEvent(Container.java:2239)
2025-03-04T09:41:06.066908 tid139960456754944 [IFO,App]   at java.awt.Component.dispatchEventImpl(Component.java:4889)
2025-03-04T09:41:06.066912 tid139960456754944 [IFO,App]   at java.awt.Container.dispatchEventImpl(Container.java:2297)
2025-03-04T09:41:06.066916 tid139960456754944 [IFO,App]   at java.awt.Component.dispatchEvent(Component.java:4711)
2025-03-04T09:41:06.066920 tid139960456754944 [IFO,App]   at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
2025-03-04T09:41:06.066924 tid139960456754944 [IFO,App]   at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
2025-03-04T09:41:06.066928 tid139960456754944 [IFO,App]   at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
2025-03-04T09:41:06.066931 tid139960456754944 [IFO,App]   at java.awt.Container.dispatchEventImpl(Container.java:2283)
2025-03-04T09:41:06.066935 tid139960456754944 [IFO,App]   at java.awt.Window.dispatchEventImpl(Window.java:2746)
2025-03-04T09:41:06.066939 tid139960456754944 [IFO,App]   at java.awt.Component.dispatchEvent(Component.java:4711)
2025-03-04T09:41:06.066943 tid139960456754944 [IFO,App]   at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
2025-03-04T09:41:06.066950 tid139960456754944 [IFO,App]   at java.awt.EventQueue.access$500(EventQueue.java:97)
2025-03-04T09:41:06.066954 tid139960456754944 [IFO,App]   at java.awt.EventQueue$3.run(EventQueue.java:709)
2025-03-04T09:41:06.066958 tid139960456754944 [IFO,App]   at java.awt.EventQueue$3.run(EventQueue.java:703)
2025-03-04T09:41:06.066962 tid139960456754944 [IFO,App]   at java.security.AccessController.doPrivileged(Native Method)
2025-03-04T09:41:06.066966 tid139960456754944 [IFO,App]   at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
2025-03-04T09:41:06.066970 tid139960456754944 [IFO,App]   at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
2025-03-04T09:41:06.066974 tid139960456754944 [IFO,App]   at java.awt.EventQueue$4.run(EventQueue.java:733)
2025-03-04T09:41:06.066978 tid139960456754944 [IFO,App]   at java.awt.EventQueue$4.run(EventQueue.java:731)
2025-03-04T09:41:06.066981 tid139960456754944 [IFO,App]   at java.security.AccessController.doPrivileged(Native Method)
2025-03-04T09:41:06.066985 tid139960456754944 [IFO,App]   at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
2025-03-04T09:41:06.067001 tid139960456754944 [IFO,App]   at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
2025-03-04T09:41:06.067011 tid139960456754944 [IFO,App]   at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
2025-03-04T09:41:06.067015 tid139960456754944 [IFO,App]   at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
2025-03-04T09:41:06.067019 tid139960456754944 [IFO,App]   at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
2025-03-04T09:41:06.067023 tid139960456754944 [IFO,App]   at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
2025-03-04T09:41:06.067027 tid139960456754944 [IFO,App]   at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
2025-03-04T09:41:06.067031 tid139960456754944 [IFO,App]   at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

Additionally, I encountered an error during installation/build time, though I’m not sure if it’s related. Here’s the error from the build process:

[... 20 lines cut ...]
 [mm-javac]                                         ^
 [mm-javac] /micro-manager/plugins/SharpnessInspector/src/main/java/org/micromanager/sharpnessinspector/SharpnessInspectorController.java:180: warning: [deprecation] getIsLiveModeOn() in SnapLiveManager has been deprecated
[... cut ...]
 [mm-javac] /micro-manager/plugins/SharpnessInspector/src/main/java/org/micromanager/sharpnessinspector/SharpnessInspectorController.java:189: warning: [deprecation] getImagePlus() in DisplayWindow has been deprecated
 [mm-javac]                   ((DisplayWindow) viewer_).getImagePlus().getHeight());
 [mm-javac]                                            ^
 [mm-javac] 9 warnings

SharpnessInspector.jar:
      [jar] Building jar: /micro-manager/plugins/SharpnessInspector.jar
```

Could someone help me understand what is wrong?

Thanks in advance!

nicost, 2025-04-01:
Looks like the jars in the libraries directory did not build and/or where not on the classpath.

AndriiPidlisnyi, 2025-04-02:
@nicost 
Hi!
Thanks for the answer!
I don't know Java, so this might be a silly question.
What would be your advice on how to verify what's wrong in this case and how to fix it?

AndriiPidlisnyi, 2025-04-02:
> Which Java version are you using to run the code?

I use Java 1.8.0_442 

> Can you delete SharpnessInspector.jar from your mmplugins directory (after building and installing the code)? I doubt that you need this panel, and removing that jar from mmplugins (or wherever it is installed on linux) should avoid the error.

It worked. I deleted SharpnessInspector.ja from the mmplugins directory, and the inspector window started opening. 
Thank you very much!

closed 2025-04-17 by nicost