micro-manager/micro-manager #2110: When MDA with Z-stack is aborted, stage returns to 0 not to the previous rest position

aandreev0, 2025-05-16:
How to reproduce: 

1. Using Demo configuration move stage to position (say, 55um)
2. launch MDA dialog and set up Z stack experiment with relative Z scan from 0 to 100um, step 1um
3. pick exposure ~100ms
4. start MDA
5. while MDA is still running, abort MDA
6. Now the stage position will be 0um and not 55um as expected

Importantly, if MDA ends properly (not aborted) then stage returns to previous defined position (55um)

I tried digging around code to find where this happens, but was not very successful yet.

marktsuchida, 2025-05-16:
Thank you for reporting this! What is your setting for Tools > Options > Use new Acquisition Engine?

aandreev0, 2025-05-20:
On my setup (compiled from latest source) this bug  happens when using the new acquisition engine, but not the old one

nicost, 2025-05-29:
I can not reproduce this. Following the steps you describe (using the new acquisition engine) the Z stage returns to 55 um rather than 0um.  I also toggled the property Z-UseSequences between Yes and No, but that did not change the behavior for me.  Nevertheless, Tanner mentioned the same bug using real hardware, so I would love to be able to reproduce this with the demo config.  I also have faint recollection of already looking at this (and possibly fixing it?).  Could you try again with the latest code?

nicost, 2025-06-30:
Closing as it can not be reproduced.  Please re-open if it still does not work for you.