micro-manager/micro-manager #2161: Build an existing device adapter in source code

asincerity, 2025-06-29:
I need to modify an existing device adapter (Arduino) to add new features. I've set up Microsoft Visual Studio, Java, and other environments on my computer. After modifying the C++ code, I generated a new DLL file using micromanager.sln in Visual Studio. Then, I installed the Micro-Manager software and replaced the original mmgr_dal_Arduino with the newly generated one. On my computer, Micro-Manager runs successfully, and I can find the modified device. However, when I switch to another computer, install Micro-Manager, and replace the DLL file in the same way, the device shows as unavailable when I open the Hardware Configuration Wizard.

Could you please help with the following questions:

Is this because I haven't set up the environment properly on the other computer?
I noticed that the new DLL file is much larger than the old one. What could be the reason for this?
How can I more quickly ensure that this change works on different computers?

nicost, 2025-06-29:
Since your DLL is much larger than the old one, you most likely build a Debug version.  It does not work on the other computer since that one does not have the Debug runtimes installed.  Switch to Release, build, and try that one on the other computer.  

If your changes are backwards compatible with the previous version of the Arduino adapter, please open a Pull Request on https://github.com/micro-manager/mmCoreAndDevices, and we can incorporate your changes in the build.

asincerity, 2025-06-30:
Thank you for your suggestion. After switching to the Release configuration, it worked correctly.