micro-manager/micro-manager #2168: Raspberry Pi 5 Build (Success) and PVCam Install (Issues)

haller2016, 2025-07-13:
Hello and thank you to the team for this amazing project. I recently acquired a Raspberry Pi 5 and thought I would go down the rabbit hole of seeing if a.) micromanager could be installed and b.) if the pcie port on the Pi could work with a PCIE firewire adapter. This would enable the use of older firewire cameras like the Photometrics Coolsnap EZ that I have, which I think would be really cool. I know there was attempt a while back on a [Pi 3 model B](https://kmdouglass.github.io/posts/micro-manager-on-the-raspberry-pi/), but I thought I would give it a try on the faster Pi 5.

I am pleased to report that I was able to build micromanager on the Pi 5 by doing the following (this is similar to [build instructions](https://github.com/micro-manager/micro-manager/blob/main/doc/how-to-build.md), but I took different steps to install both JDK 8 and swig 3.0). 

`sudo apt install git subversion build-essential autoconf automake libtool pkg-config autoconf-archive ant libboost-all-dev libpcre3-dev libpcre2-dev`

I then downloaded JDK 8 from oracle, [specifically the aarch64 version](https://www.oracle.com/java/technologies/downloads/). After this, I followed [this advice and was able to get JDK recognized system wide.](https://forums.raspberrypi.com/viewtopic.php?t=231460) 

`sudo mkdir /usr/java
cd /usr/java
sudo tar xf ~/Downloads/jdk-8u451-linux-aarch64.tar.gz
sudo update-alternatives --install /usr/bin/java java /usr/java/jdk1.8.0_451/bin/java 1000
sudo update-alternatives --install /usr/bin/javac javac /usr/java/jdk1.8.0_451/bin/javac 1000
java -version`

I then downloaded swig 3.0.12 from the swig source forge (https://sourceforge.net/projects/swig/files/) and did the following steps (I included the libpcre3-dev and libpcre2-dev in the first build step so you shouldn't have to do that here!):

 `cd ~/Downloads
tar -xvzf swig-3.0.12.tar.gz
cd swig-3.0.12/
./configure
make
sudo make install`

I then proceeded with the usual steps to build micromanager on Linux. 

`git clone https://github.com/micro-manager/micro-manager.git
cd micro-manager
git submodule update --init --recursive

./autogen.sh
./configure
mkdir ../3rdpartypublic; pushd ../3rdpartypublic
svn checkout https://svn.micro-manager.org/3rdpartypublic/classext
popd
make fetchdeps
make -j4
sudo make install`

Now for the part I am a little stuck on. I want to install PVCam so that I can test whether the PCIe adapter and Coolsnap EZ will work on the Pi 5 with micromanager. I downloaded PVCam 3.10.0.3 Linux from the Teledyne website, and unzipped the .zip file. I then proceeded to run `bash pvcam__install_helper-Ubuntu.sh` in the `pvcam` folder, as instructed in the docs. It seems to install (sort of), but I get the following right at the end. 

`Building PCIe driver module for all kernels, please wait...

[ERROR] The PCIe driver build failed. PCIe cameras will not work! 
See details in '/tmp/pvcamPCIE.build-ivxXVy.log' file. 
Run the script '/opt/pvcam/drivers/in-kernel/pcie/build_and_install.sh' again once the problem is fixed.

Installation has completed successfully.
However, see the installer output for possible issues.

Installation has completed successfully.
Don't forget to reboot or re-login!`

The output of that build file is the following, and this is really where I have no idea what I am looking at. Any advice here would be really appreciated!!

`Hooking PCIe driver sources to DKMS
Re-building PCIe driver for all kernels in DKMS
Error! The module/version combo: pvcam_pcie-6.1.0 is not located in the DKMS tree.
Reading a list of installed kernel versions from '/usr/lib/modules' folder
Installing the driver for kernel 6.12.34+rpt-rpi-v8
Sign command: /lib/modules/6.12.34+rpt-rpi-v8/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/pvcam_pcie/6.1.0/source -> /usr/src/pvcam_pcie-6.1.0

Building module:
Cleaning build area...
'make' KDIR=/lib/modules/6.12.34+rpt-rpi-v8/build modules...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.12.34+rpt-rpi-v8 (aarch64)
Consult /var/lib/dkms/pvcam_pcie/6.1.0/build/make.log for more information.
Installing the driver for kernel 6.1.0-37-arm64
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' KDIR=/lib/modules/6.1.0-37-arm64/build modules.....
Signing module /var/lib/dkms/pvcam_pcie/6.1.0/build/pvcam_pcie/pvcam_pcie.ko
Cleaning build area...

pvcam_pcie.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.1.0-37-arm64/updates/dkms/
depmod...
Installing the driver for kernel 6.12.25+rpt-rpi-v8
Sign command: /lib/modules/6.12.25+rpt-rpi-v8/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' KDIR=/lib/modules/6.12.25+rpt-rpi-v8/build modules...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.12.25+rpt-rpi-v8 (aarch64)
Consult /var/lib/dkms/pvcam_pcie/6.1.0/build/make.log for more information.
Installing the driver for kernel 6.12.25+rpt-rpi-2712
Sign command: /lib/modules/6.12.25+rpt-rpi-2712/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' KDIR=/lib/modules/6.12.25+rpt-rpi-2712/build modules...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.12.25+rpt-rpi-2712 (aarch64)
Consult /var/lib/dkms/pvcam_pcie/6.1.0/build/make.log for more information.
Installing the driver for kernel 6.12.34+rpt-rpi-2712
Sign command: /lib/modules/6.12.34+rpt-rpi-2712/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' KDIR=/lib/modules/6.12.34+rpt-rpi-2712/build modules...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.12.34+rpt-rpi-2712 (aarch64)
Consult /var/lib/dkms/pvcam_pcie/6.1.0/build/make.log for more information.
(Re)loading PCIe driver
modprobe: FATAL: Module pvcam_pcie not found in directory /lib/modules/6.12.34+rpt-rpi-2712
[ERROR] at "/opt/pvcam/drivers/in-kernel/pcie/build_and_install.sh":100 command: modprobe "pvcam_pcie"
`

haller2016, 2025-07-14:
So the good news is that the video4linux does seem to work well with USB devices, or at least with the Logitech webcam I had lying around. Both 8bit and color seem to work well. The bad news is that when I went to try the OpenCVgrabber, I realized it wasn't actually in the list of devices in the hardware manager. Looking further I realized that OpenCVgrabber, IIDC (firewire), and PVCam were all missing. I don't remember seeing these as errors when building micromanager, but is it possible that they didn't build properly? When I look into the clone I made (specifically in the ~/micro-manager/mmCoreAndDevices/DeviceAdapters directory) I do see them. In the meantime, I will see if the HQ camera works with video4linux.

marktsuchida, 2025-07-15:
The `./configure` script will skip (exclude from the build) all device adapters that depend on libraries that cannot be found. There are _lots_ of device adapters that require an external library, some of which may not even work on Linux (or arm64), so I would recommend concentrating only on the ones you need/want.

In theory OpenCV, libdc1394, etc., should be automatically detected if installed, possibly with the help of extra flags or variables to `./configure` (see `./configure --help=recursive`).

The OpenCVgrabber device adapter is currently written against an old version (2.4.x) of OpenCV, so some trial and error may be needed to get it to build -- you can either manually build and install an old version, or try and see if you can adapt the device adapter code to a current version. If building an old version, the [macOS build script](https://github.com/micro-manager/micro-manager/blob/b6ec4e078c4c40bed3b6ae8912a8346f49f44cc2/buildscripts/nightly/nightlybuild_macOS_deps.sh#L458) (also [here](https://github.com/micro-manager/micro-manager/blob/b6ec4e078c4c40bed3b6ae8912a8346f49f44cc2/buildscripts/nightly/nightlybuild_macOS_mm.sh#L90)) might provide some hints, though they also contain macOS-specific stuff.

haller2016, 2025-07-15:
Some more good news. I have decided to go through and test a few that I think I could get working (or at least show up in the Hardware Configuration). **_I will come back to this post and update the list as I test more and more_**.  

1. video4linux (confirmed working with Logitech webcam)
2. IIDC and dc1394 (now show up after install via ```libdc1394-dev```). I think I can confirm/deny whether this works for sure with a legacy AlliedVision camera in the next few days. 
3. Basler Pylon (7.1.0 as stated by the docs). Accidentally installed the latest version but then realized the docs specifically called for 7.1.0. The Basler Ace we currently have in the lab is in an experiment so it may take me some time to truly test this. 

One that I am attempting to get working but is being fussy is gphoto2/libgphoto2 library. I installed gphoto2 simply with ```sudo apt install gphoto2``` and confirmed installation with ```gphoto --version```, which gives: 

```
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.30         standard camlibs, gcc, no ltdl, EXIF
libgphoto2_port 0.12.1         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking
```

However when I configure with ```./configure --with-gphoto2=yes```, it seems to be mad at me. 

```
checking if libgphoto2 support was requested... yes
checking for user-provided configuration for libgphoto2... no; will perform availability check
checking for pkg-config... (cached) /usr/bin/pkg-config
checking pkg-config is at least version 0.20... yes
checking for pkg-config metadata for libgphoto2 >= 2.5.1... no
checking for gphoto2/gphoto2.h with hard-coded flags... no
configure: error: in `/home/eric/micro-manager/mmCoreAndDevices/DeviceAdapters':
configure: error: --with-gphoto2 was given, but test for libgphoto2 failed
See `config.log' for more details
configure: error: ./configure failed for mmCoreAndDevices/DeviceAdapters
```

I am realizing this is quickly turning into me testing as many devices as possible, which I am happy to do haha. I will update the name of the post accordingly. I had a friend of mine look over the PVCam logs and they also suggested a kernel downgrade may be necessary, so I may table that one for now while I keep testing the easy (or easier) ones. 

-Eric

benlongo, 2025-07-16:
You probably need `libgphoto2-dev` to compile against.