micro-manager/micro-manager #2184: Crash on Windows in Java 11 code

nicost, 2025-08-13:
We experienced Micro-Manager crashing (as in disappearing from the screen) at random moments.  In some cases, the screen went black for a while and then came back on (it was hard to tell if this was a full reboot of the computer since this PC auto-logs into the account.  I obtained a crash resistant report and found the following.  I believe this to be an exception inside the Java implementation, triggered by one of the Micro-Manager callbacks.  I will trace the callbacks to see if one of them could result in rendering something not on the EDT.  It is interesting this apparently only happens on this PC (which runs Windows 10).

```
2025-08-12T17:34:55.808060 tid6020 [dbg,dev:TIScope] [TiCOMPFS] Invoking notification callback for enabled (0)
2025-08-12T17:34:55.808259 tid6020 [dbg,App] Notification for Device: TIPFSOffset Property: Position changed to value: 60
2025-08-12T17:35:00.834212 tid6020 [dbg,dev:TIScope] [TiCOMNosepiece] Invoking notification callback for position (0)
2025-08-12T17:35:01.342861 tid6020 [dbg,dev:TIScope] [TiCOMNosepiece] Invoking notification callback for position (0)
2025-08-12T17:35:01.357819 tid6020 [dbg,dev:TIScope] [TiCOMNosepiece] Invoking notification callback for position (0)
2025-08-12T17:35:01.722909 tid6020 [dbg,dev:TIScope] [TiCOMNosepiece] Invoking notification callback for position (3)
2025-08-12T17:35:01.723046 tid6020 [dbg,App] Notification for Device: TINosePiece Property: State changed to value: 2
2025-08-12T17:35:01.723245 tid6020 [dbg,App] Notification for Device: TINosePiece Property: Label changed to value: 3-Plan Fluor 40x NA 1.30 Oil
2025-08-12T17:35:44.338380 tid6020 [dbg,dev:TIScope] [TiCOMPFS] Invoking notification callback for status (3)
2025-08-12T17:35:44.338478 tid6020 [dbg,App] Notification for Device: TIPFSStatus Property: Status changed to value: Dichroic mirror not inserted
2025-08-12T17:35:44.338671 tid6020 [dbg,dev:TIScope] [TiCOMPFS] Invoking notification callback for enabled (0)
2025-08-12T17:35:46.295350 tid6020 [dbg,dev:TIScope] [TiCOMPFS] Invoking notification callback for status (1)
2025-08-12T17:35:46.295489 tid6020 [dbg,App] Notification for Device: TIPFSStatus Property: Status changed to value: Out of focus search range
2025-08-12T17:35:46.295706 tid6020 [dbg,dev:TIScope] [TiCOMPFS] Invoking notification callback for enabled (0)
2025-08-12T17:35:47.345396 tid6020 [dbg,dev:TIScope] [TiCOMPFS] Invoking notification callback for status (3)
2025-08-12T17:35:47.345494 tid6020 [dbg,App] Notification for Device: TIPFSStatus Property: Status changed to value: Dichroic mirror not inserted
2025-08-12T17:35:47.345689 tid6020 [dbg,dev:TIScope] [TiCOMPFS] Invoking notification callback for enabled (0)
***** END Captured Log *****
***** BEGIN HotSpot Error Log (C:\Program Files\Micro-Manager-2.0\hs_err_pid11064.log) *****
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000001e178322483, pid=11064, tid=11232
#
# JRE version: OpenJDK Runtime Environment Temurin-11.0.27+6 (11.0.27+6) (build 11.0.27+6)
# Java VM: OpenJDK 64-Bit Server VM Temurin-11.0.27+6 (11.0.27+6, mixed mode, tiered, compressed oops, g1 gc, windows-amd64)
# Problematic frame:
# J 12126 c2 com.sun.java.swing.plaf.windows.WindowsButtonUI.paint(Ljava/awt/Graphics;Ljavax/swing/JComponent;)V java.desktop@11.0.27 (18 bytes) @ 0x000001e178322483 [0x000001e178320be0+0x00000000000018a3]
#
# No core dump will be written. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   https://github.com/adoptium/adoptium-support/issues
#

---------------  S U M M A R Y ------------

Command Line: -XX:SoftRefLRUPolicyMSPerMB=1000000000000 -Xmx2000m -XX:MaxDirectMemorySize=1000g -XX:+UseG1GC ij.ImageJ -ijpath C:\PROGRA~1\MICRO-~1.0\

Host: Intel(R) Core(TM) i9-10900X CPU @ 3.70GHz, 20 cores, 127G,  Windows 10 , 64 bit Build 19041 (10.0.19041.5915)
Time: Tue Aug 12 17:35:58 2025 Pacific Daylight Time elapsed time: 1061.119809 seconds (0d 0h 17m 41s)

---------------  T H R E A D  ---------------

Current thread (0x000001e10993b000):  JavaThread "AWT-EventQueue-0" [_thread_in_Java, id=11232, stack(0x000000b528300000,0x000000b528400000)]

Stack: [0x000000b528300000,0x000000b528400000],  sp=0x000000b5283fba80,  free space=1006k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  0x000001e178322483

siginfo: EXCEPTION_ACCESS_VIOLATION (0xc0000005), reading address 0x0000000000000020

Register to memory mapping:

RIP=0x000001e178322483 is at entry_point+6307 in (nmethod*)0x000001e178320310
RAX=0x0 is NULL
RBX=0x000000000000003e is an unknown value
RCX=0x0000000000000009 is an unknown value
RDX=0x0 is NULL
RSP=0x000000b5283fba80 is pointing into the stack for thread: 0x000001e10993b000
RBP=0x0000000000000020 is an unknown value
RSI=0x00000000830a6c98 is an oop: [B 
{0x00000000830a6c98} - klass: {type array byte}
 - length: 9
RDI=0x0000000000000029 is an unknown value
R8 =0x0000000000000009 is an unknown value
R9 =0x0000000000000029 is an unknown value
[... 120 lines cut ...]
                                   [       ]   sun.cpu.endian = little
                                   [       ]   sun.cpu.isalist = amd64
                                   [       ]   sun.desktop = windows
                                   [       ]   sun.io.unicode.encoding = UnicodeLittle
                                   [       ]   sun.java.command = ij.ImageJ -ijpath C:\PROGRA~1\MICRO-~1.0\
                                   [       ]   sun.java.launcher = SUN_STANDARD
                                   [       ]   sun.jnu.encoding = Cp1252
                                   [       ]   sun.management.compiler = HotSpot 64-Bit Tiered Compilers
                                   [       ]   sun.os.patch.level = 
                                   [       ]   user.country = US
                                   [       ]   user.dir = C:\Program Files\Micro-Manager-2.0
                                   [       ]   user.home = C:\Users\HHMI
                                   [       ]   user.language = en
                                   [       ]   user.name = HHMI
                                   [       ]   user.script = 
                                   [       ]   user.timezone = America/Los_Angeles
                                   [       ]   user.variant = 
                                   [       ] (End Java system properties)
2025-08-12T17:18:50.419656 tid7868 [IFO,App] Processors available to JVM: 20
2025-08-12T17:18:50.420687 tid7868 [IFO,App] Total physical memory (caveats apply if JVM is 32-bit): 137092694016 (127.7 GiB)
                                   [       ] Committed virtual memory size: 3794849792 (3.5 GiB)
                                   [       ] Free physical memory (may be meaningless if JVM is 32-bit): 130690314240 (121.7 GiB)
2025-08-12T17:18:50.421429 tid7868 [IFO,App] JVM heap memory usage: used = 116492416 (111.1 MiB); committed = 2097152000 (2.0 GiB); max = 2097152000 (2.0 GiB)
                                   [       ] JVM non-heap memory usage: used = 182421504 (174.0 MiB); committed = 185860096 (177.2 MiB); max = unavailable
```

marktsuchida, 2025-08-14:
Hmm, it does look like the crash is happening on the EDT (tid 11232), though.

> In some cases, the screen went black for a while and then came back on

Could also be something related to GPU drivers, if other avenues don't lead to a solution. Just an idea.

nicost, 2025-08-15:
It currently looks like this was do to a BIOS setting problem, and I am now running a long test after resetting the CMOS o nthe motherboard.  Likely will also need a new battery.  Weird issue.,,,

I did work a bit on making sure that callbacks run on the EDT in relevant sections.  I will certainly hold of on merging that.  It may be good to diccuss a good strategy for callback thread management first.

nicost, 2025-08-17:
Indeed. BIOS issue. Closing.

closed 2025-08-17 by nicost