micro-manager/micro-manager #2190: Zeiss MCU28 control of X_Y stage on Observer Z1

Jez-CustomScopes, 2025-08-20:
I am able to control the microscope body using the CAN29 device adapter with a null modem usb adapter to the microscope main body.  However in order control the X_Y stage I believe that I need to connect the MCU28 to the CAN device adapter. This connects OK, also using a null modem USB to serial adapter, but at 9600 baud rather tnan 57600 baud.

[image]

but when I save the config it returns the following message, presumably when it initialises. (line 31)

[image]

Jez-CustomScopes, 2025-08-27:
> Am I understanding right that the microscope and the MCU28 are on two different serial ports, one set to 57600 baud and the other to 9600? Which one is COM6?
>
> Also, do the two devices individually work if you create a configuration containing just one or the other?
>
> In any case, it might be helpful if you could capture a debug log using the Problem Report (in Help menu; start it before opening the Hardware Configuration Wizard).

Thanks Mark,
I have added Neelima, the end-user, to the email thread.
Yes, we tested the MCU28 separately (without the microscope body connected) on two different USB to serial adapters (both null modem) and it still won't initialise, despite seeming to connect (if the baud rate isn't 9600 then it doesn't get to this point). The screenshots were taken with it connected to COM6 at 9600 Baud.
@Neelima, please could you capture a debug log for Mark

https://micro-manager.org/Micro-Manager_FAQ-new

Best wishes,
Jeremy

Jez Graham - Custom Scopes

marktsuchida, 2025-08-27:
For the MCU28, in addition to setting 9600 baud, make sure to try also setting Handshaking to Hardware and AnswerTimeout to 2000 (ms) -- if you haven't already. The rest should be the defaults (8 data bits, 1 stop bit, no parity).

Jez-CustomScopes, 2025-08-28:
Hi Mark,
We already set the timeout to 2000 ms, I'm pretty sure I tried longer than this too, but will test again.
BW,
Jez

Jez Graham - Custom Scopes

Jez-CustomScopes, 2025-09-19:
The USB to Null modem adapter worked fine with the Zeiss body on CAN29, but not with the MCU28 on CAN.  We have subsequently ordered and tried a different adapter with all the settings you suggested and it WORKED.
Link to device below:  
[https://www.amazon.co.uk/dp/B08T19WFCC?ref=fed_asin_title&th=1](https://urldefense.proofpoint.com/v2/url?u=https-3A__www.amazon.co.uk_dp_B08T19WFCC-3Fref-3Dfed-5Fasin-5Ftitle-26th-3D1&d=DwMF-g&c=euGZstcaTDllvimEN8b7jXrwqOf-v5A_CdpgnVfiiMM&r=raANmoGPxkOgpsOWcsTzMiEaMruRY8TfW6Ul_0Ge4Ds&m=34jxT-kwKbEFZ67S0KkaK-xDE26-vvWbNeH5Nr6EmJv5EIJbimmtqWTUzEuWenKd&s=_jP1MuOIX2b5jdC8IT3t76Dph79Vl1fD2q2YhFgxgmA&e=)

Green-utech USB to RS232 DB9 Female Null Modem 2-TXD Rollover Serial Cable

Thanks for the help.

closed 2025-09-19 by marktsuchida