micro-manager/micro-manager #2201: Tile Creator doesn't account for binning

jondaniels, 2025-09-16:
Tile Creator within the Position List apparently determines the stage increment in X and Y according to the number of pixels and the pixel size and does not correctly account for binning.

Easy way to reproduce using the demo config: move +/- 1 field in X and Y and set those in the grid creator with binning set to 2.  When the position list is created there will be 16 positions instead of 4.

nicost, 2025-10-17:
I can not reproduce this in the current code:

[image]

nicost, 2025-10-20:
I believe we work off the same code.  I followed the steps you describe.  However, it is highly likely that the exact order of things (changing the pixel size in the Tile Creator window, when to press "Center Here", versus changing binning) is important.  
Looking at the dialog now, I'd argue that Pixel Size should be removed (as it always should be read from the configuration). Funny thing: in my example above, pixel size in the Tile Creator Dialog is wrong.